libts/tslib #25: Is it possible to add a Android.mk file

gaara92, 2015-02-27:
Hi, 
I would like to build tslib at the same time as Android, (eGalax issue & Android 4.2)
Is it possible to add this?

gaara92, 2015-03-05:
I have found an Android file here: https://github.com/etmatrix/tslib/blob/master/Android.mk
I have juste replace all the
LOCAL_MODULE_TAGS := optional
by:
LOCAL_MODULE_TAGS := eng
And add the folder in /android/external/ folder before build the image.

Now, I can run ts_calibrate (without GUI) and I can get a pointercal file.
But I can't use it for the moment.

kergoth, 2015-03-05:
This really isn't the responsibility of an upstream project like this. Android generally takes care of integration of upstream projects themselves.

closed 2015-03-05 by kergoth