libts/tslib #28: Daemon which translates tslib events to uinput

vpeter4, 2015-06-29:
Recently I need to control Kodi (Open Source Home Theatre Software) on OpenELEC Linux distro with touchscreen. Because Kodi doesn't support touchcreens I was looking around what to do. I come across uinput-mapper which is python script. But seems it has some issues. So I wrote one small application which translates tslib events to uinput. And it works very well. If this would be useful for official tslib let me know and I will make PR. Or if anyone has any suggestion about this let me know.

My work is here: https://github.com/vpeter4/tslib/commit/2ae91258548c3682ad5df16ee22bdf28d9769bfe

merge, 2016-11-14:
just in case anyone is wondering: such a program, only a little more generic and hopefully usable for basically any device is in the works and planned to be included in tslib 1.3 (next year).

vpeter4, 2016-11-14:
Little off-topic but who is using tslib this days? For which projects?

merge, 2016-11-14:
not many do, I think. Touchscreen chips can apply the necessary filters too these days. In my case, we have users who just insist on having the possibility to use the filters and manual calibration, even for newer devices, just because they always had them.

Technically, tslib's benefit is to apply filters and calibration in software and to change them during runtime; or to have a single place to use filters independently from the device.

One main problem though is exactly this issue; imo, nobody should need to understand tslib. tslib should just become a normal input device that behaves like your original one, with filters applied. I can imagine tslib getting users once we have this and multitouch, both scheduled for version 1.3.

merge, 2016-11-23:
said program is merged and as of now developed in master directly, see tools/ts_uinput.c

closed 2016-11-23 by merge

vpeter4, 2016-11-23:
Very, very nice! Will test it when time permits.