libts/tslib #29: How to exit ts_test

cefn, 2015-07-20:
I was following some instructions to run ts_test as part of the configuration of a PiTFT module from Adafruit as described here...
https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi

I ran...

    sudo TSLIB_FBDEVICE=/dev/fb1 TSLIB_TSDEVICE=/dev/input/touchscreen ts_test

...and I now have a fullscreen application with no window controls and no mechanism for exiting the test program. 

Looks like I have to hard de-power my Raspberry Pi to be able exit the ts_test program. Have I missed something. Is this intended? 

What other way of exiting the program is there? I have tried Esc, CTRL-Q CTRL-ALT-Backspace CTRL+Break CTRL+C, CTRL+Z

kergoth, 2015-07-20:
ts_test creates Draw, Drag, and Quit buttons on the framebuffer. Assuming a functional touchscreen, you can click on quit. If the touchscreen is not functional, and your terminal is functional, then ctrl+c should work, as far as I know, as that will sent a INT signal to the process. You can also use `kill` or `killall` or `pkill` from another terminal (e.g. ssh, or another virtual terminal) to exit the process via the TERM signal.

cefn, 2015-07-20:
Thanks for your reply.

The version I was running had no visible or accessible Quit button, (possibly due to low resolution - 480x320)? The other buttons you mentioned were there.

CTRL+C didn't function to close down the program, and it seemed to be capturing all Key events so the VT switch which I could previously trigger through CTRL+ALT+F2 (when in the console) was non-functional. 

I could see the previously-switched terminal still running on the HDMI-attached screen, but not get focus to it. If I'd managed to get focus there I'm pretty sure I could have killed the program.

willwac, 2015-09-20:
I would recommend placing the quit button in a different position or allowing you to focus with CTRK-ALT-F1-6

merge, 2016-11-18:
Ctrl-c works if you have a terminal. As kergoth said, if not, you can kill it as long as you can reach the device. I don't really see any reason for action here, but what would be a better position for the quit button, in case the user really somehow can't exit? according to @cefn swithing place with "Drag" would help here. If this is still a problem for anyone here, I could imange doing this, so please tell me. Otherwise I'll keep waiting a while before I close this issue.

merge, 2016-11-23:
Given this issue's age and that the bug wasn't really reproduced, I'll go forth and close here. Please feel free to open a new issue though in case you have wishes for ts_test in tslib's current version.

williamhatcher, 2017-01-21:
I would recommend putting the quit button in between the other two buttons, making them a tad smaller.

merge, 2017-01-24:
I hesitate a bit to change the buttons much; who knows what people are doing with ts_test. I think *adding* an X or whatever in the upper left corner would most probably help without breaking things.

willwac, 2017-01-24:
Yes that would, however, be careful about cluttering up that area. Maybe
put it in the bottom corner?

merge, 2017-01-24:
True if 0 is at the bottom. We'd try to cover low resolution problems...

merge, 2017-02-24:
Right now, I don't really want to change ts_test. As for ts_test_mt, it would be cool to have an "SDL mode" in order to make it perfectly usable in graphical environments. That way, exiting wouldn't be an issue.

0, in my case, is at the top left. If there isn't more pressure, it's 50:50 chance whether I'll add an X in ts_test_mt there someday :)

willwac, 2017-02-24:
I haven't used ts_test_mt, so this issue can probably be closed.

However, I do want to point out that Adafruit's article still recommends
using ts_test to test their TFTs.

closed 2017-03-12 by merge