libts/tslib #30: can't compile tslib on ubuntu 14.04

flashsouth, 2015-08-20:
hello, 

i've been playing around with the Beaglebone black , and before i succeded to compile tslib, but after an unknown update i can't compile it anymore,

here's the output when i'm trying to execute the shell script autogen.sh : 

libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in `m4/internal'.
libtoolize: copying file `m4/internal/libtool.m4'
libtoolize: copying file `m4/internal/ltoptions.m4'
libtoolize: copying file `m4/internal/ltsugar.m4'
libtoolize: copying file `m4/internal/ltversion.m4'
libtoolize: copying file `m4/internal/lt~obsolete.m4'
libtoolize: Consider adding `AC_CONFIG_MACRO_DIR([m4])' to configure.ac and
libtoolize: rerunning libtoolize, to keep the correct libtool macros in-tree.
configure.ac:27: installing './config.guess'
configure.ac:27: installing './config.sub'
configure.ac:7: installing './install-sh'
configure.ac:7: installing './missing'
plugins/Makefile.am:13: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am:23: warning: source file '$(top_srcdir)/plugins/linear.c' is in a subdirectory,
src/Makefile.am:23: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
src/Makefile.am:27: warning: source file '$(top_srcdir)/plugins/dejitter.c' is in a subdirectory,
src/Makefile.am:27: but option 'subdir-objects' is disabled
src/Makefile.am:31: warning: source file '$(top_srcdir)/plugins/linear-h2200.c' is in a subdirectory,
src/Makefile.am:31: but option 'subdir-objects' is disabled
src/Makefile.am:35: warning: source file '$(top_srcdir)/plugins/variance.c' is in a subdirectory,
src/Makefile.am:35: but option 'subdir-objects' is disabled
src/Makefile.am:39: warning: source file '$(top_srcdir)/plugins/pthres.c' is in a subdirectory,
src/Makefile.am:39: but option 'subdir-objects' is disabled
src/Makefile.am:43: warning: source file '$(top_srcdir)/plugins/ucb1x00-raw.c' is in a subdirectory,
src/Makefile.am:43: but option 'subdir-objects' is disabled
src/Makefile.am:47: warning: source file '$(top_srcdir)/plugins/corgi-raw.c' is in a subdirectory,
src/Makefile.am:47: but option 'subdir-objects' is disabled
src/Makefile.am:51: warning: source file '$(top_srcdir)/plugins/collie-raw.c' is in a subdirectory,
src/Makefile.am:51: but option 'subdir-objects' is disabled
src/Makefile.am:55: warning: source file '$(top_srcdir)/plugins/h3600-raw.c' is in a subdirectory,
src/Makefile.am:55: but option 'subdir-objects' is disabled
src/Makefile.am:59: warning: source file '$(top_srcdir)/plugins/mk712-raw.c' is in a subdirectory,
src/Makefile.am:59: but option 'subdir-objects' is disabled
src/Makefile.am:63: warning: source file '$(top_srcdir)/plugins/arctic2-raw.c' is in a subdirectory,
src/Makefile.am:63: but option 'subdir-objects' is disabled
src/Makefile.am:67: warning: source file '$(top_srcdir)/plugins/tatung-raw.c' is in a subdirectory,
src/Makefile.am:67: but option 'subdir-objects' is disabled
src/Makefile.am:71: warning: source file '$(top_srcdir)/plugins/input-raw.c' is in a subdirectory,
src/Makefile.am:71: but option 'subdir-objects' is disabled
tests/Makefile.am:12: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')

Thanks in advance

closed 2015-08-20 by kergoth

flashsouth, 2015-08-20:
thanks for the solution , although there's a little issue after i execute ./autogen now : 
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in `m4/internal'.
libtoolize: copying file `m4/internal/libtool.m4'
libtoolize: copying file `m4/internal/ltoptions.m4'
libtoolize: copying file `m4/internal/ltsugar.m4'
libtoolize: copying file `m4/internal/ltversion.m4'
libtoolize: copying file `m4/internal/lt~obsolete.m4'
libtoolize: Consider adding `AC_CONFIG_MACRO_DIR([m4])' to configure.ac and
libtoolize: rerunning libtoolize, to keep the correct libtool macros in-tree.
configure.ac:18: installing './compile'
configure.ac:27: installing './config.guess'
configure.ac:27: installing './config.sub'
configure.ac:7: installing './install-sh'
configure.ac:7: installing './missing'
plugins/Makefile.am:13: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
plugins/Makefile.am: installing './depcomp'
tests/Makefile.am:12: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')

flashsouth, 2015-08-21:
Hello ,

Also , when i try make install i get this output at the end : 

make[2]: Nothing to be done for `install-data-am'.
make[2]: Leaving directory `/home/hicham/tslib/plugins'
make[1]: Leaving directory `/home/hicham/tslib/plugins'
Making install in tests
make[1]: Entering directory `/home/hicham/tslib/tests'
make[2]: Entering directory `/home/hicham/tslib/tests'
 /bin/mkdir -p '/home/hicham/tslib_beaglebone/bin'
  /bin/bash ../libtool   --mode=install /usr/bin/install -c ts_test ts_calibrate ts_print ts_print_raw ts_harvest '/home/hicham/tslib_beaglebone/bin'
libtool: install: /usr/bin/install -c .libs/ts_test /home/hicham/tslib_beaglebone/bin/ts_test
libtool: install: /usr/bin/install -c .libs/ts_calibrate /home/hicham/tslib_beaglebone/bin/ts_calibrate
libtool: install: /usr/bin/install -c .libs/ts_print /home/hicham/tslib_beaglebone/bin/ts_print
libtool: install: /usr/bin/install -c .libs/ts_print_raw /home/hicham/tslib_beaglebone/bin/ts_print_raw
libtool: install: /usr/bin/install -c .libs/ts_harvest /home/hicham/tslib_beaglebone/bin/ts_harvest
make[2]: Nothing to be done for `install-data-am'.
make[2]: Leaving directory `/home/hicham/tslib/tests'
make[1]: Leaving directory `/home/hicham/tslib/tests'
make[1]: Entering directory `/home/hicham/tslib'
make[2]: Entering directory `/home/hicham/tslib'
make[2]: Nothing to be done for `install-exec-am'.
 /bin/mkdir -p '/home/hicham/tslib_beaglebone/lib/pkgconfig'
 /usr/bin/install -c -m 644 tslib-1.0.pc tslib.pc '/home/hicham/tslib_beaglebone/lib/pkgconfig'
make[2]: Leaving directory `/home/hicham/tslib'
make[1]: Leaving directory `/home/hicham/tslib'

kergoth, 2015-08-21:
Those aren't errors, those are the usual messages from a make install.

commit 0a11148e (2015-08-20): configure.ac: enable subdir-objects

Fixes #30