libts/tslib #34: tslib enhancements / work without tslib.conf / multitouch-as-singletouch

AlexanderSchmidCcv, 2015-11-26:
Hello,

I have made some changes on the tslib-1.1 code and would like to post it back to you, for review and incorporating them into the main code stream. 

There are two main features which I added:
- publish the  ts_load_module_raw function on tslib API. This lets users load modules without the need of a tslib.conf file. I found this especially useful when working with capacitive touchscreens, where no calibration settings are needed. 

- Add a very basic support to handle multitouch screens, so that they report touches for the first finger as if they were single-touch devices.

From my point of view, everybody could profit of those changes, so I would be happy if they were incorporated into the main stream.
I attached the changed files (taken originally from tslib-1.1).

Regards,
Alex

[tslib.h.txt](https://github.com/kergoth/tslib/files/44882/tslib.h.txt)
[ts_load_module.c.txt](https://github.com/kergoth/tslib/files/44883/ts_load_module.c.txt)
[linear.c.txt](https://github.com/kergoth/tslib/files/44884/linear.c.txt)
[input-raw.c.txt](https://github.com/kergoth/tslib/files/44885/input-raw.c.txt)

Details per file:
1) tslib.h
- remove comma at end of ts_param enumeration, to satisfy compiler (produces error when running in pedantic mode)
- add ts_load_module_raw to the public interface so that applications can load modules without having a tslib.conf file

2) ts_load_module.c
- add the parameters to the debugging output when a module is loaded

3) linear.c
- convert %d to %u to satisfy compiler

4) input-raw.c
- Add extensions for multitouch screens to support them in a single-touch style
- Add mor debug information logging

merge, 2016-11-16:
if you fork and create small, individual patches, each doing one little thing, I might pick one or more of those up, if appropriate. Otherwise I'll do the changes myself. I hope you understand it's difficult to take them in your current form. thanks!

AlexanderSchmidCcv, 2016-11-16:
Hello,
well I´m afraid I cannot do this. I am not allowed using git/github here in my company office. This is why I provided the changed files as attachment.
I hope you can integrate the changes easily, using a diff-tool should reveal the changes.
If you need comments/explanations on my changes, feel free to contact me.
Thanks!

AlexanderSchmidCcv, 2016-11-16:
Sorry, pushed the wrong button, didn´t mean to close...

merge, 2016-11-17:
to your changes:

- tslib.h: comma removed. thanks for that. Especially for version 1.2 I don't have plans to add an ABI for users. I'll keep the idea of loading and unloading of modules though. Maybe I'll mention it in TODO

- ts_load_module.c added the parameters at the place where everyone (when using DEBUG) will see them. thanks!

- input_raw.c: there aren't unknown event types. we just don't handle them. Multitouch screens should already be supported in single touch. But multitouch is a broader topic and will be tackled for version 1.3. so here I didn't take your changes.

thanks a lot for your work!

closed 2016-11-18 by merge