libts/tslib #35: compilation problem under Debian Jessie 8.2

bremenpl, 2015-12-13:
Hello there,
I am using Debian 8.2:
```
lukasz@vm-debian:~/tslib$ uname -a
Linux vm-debian 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt11-1+deb8u3 (2015-08-04) x86_64 GNU/Linux
```

I am trying to compile tslib for usage with BeagleBone black. I have done following steps to compile tslib:

```
sudo apt-get update
sudo apt-get install autoconf libtool
git clone https://github.com/kergoth/tslib.git
```

Now I export my crosscompiller path, its visible:
```
lukasz@vm-debian:~/tslib$ arm-linux-gnueabihf-gcc
arm-linux-gnueabihf-gcc: fatal error: no input files
compilation terminated.
```

Then i start to compile:
```
./autogen.sh // ALL OK
./configure --prefix=/home/lukasz/tslib --host=arm-linux-gnueabihf // ALL OK
make // ALL OK
make install // ERROR

lukasz@vm-debian:~/tslib$ make install
Making install in etc
make[1]: Entering catalog '/home/lukasz/tslib/etc'
make[2]: Entering catalog '/home/lukasz/tslib/etc'
 /bin/mkdir -p '/home/lukasz/tslib/etc'
 /usr/bin/install -c -m 644 ts.conf '/home/lukasz/tslib/etc'
/usr/bin/install: „ts.conf” i „/home/lukasz/tslib/etc/ts.conf” is the same file
Makefile:326: commands for object 'install-sysconfDATA' failed
make[2]: *** [install-sysconfDATA] Error 1
make[2]: leaving directory '/home/lukasz/tslib/etc'
Makefile:396: commands for object 'install-am' failed
make[1]: *** [install-am] Error 2
make[1]: Leaving catalog '/home/lukasz/tslib/etc'
Makefile:467: commands for object 'install-recursive' failed
make: *** [install-recursive] Error 1
```
I translated the output from polsih i hope its consistent.
Could you please advice me what I got wrong? I would really apreciate all help.

playya, 2015-12-13:
Hi,

Your prefix equals your source directory which ends in a collision of
files.

If you want to be lazy and get generic output you can set LANG=c and
LC_ALL= using export.

Regards,
Fred

bremenpl, 2015-12-13:
I wouldnt say I want to be lazy, I just wanted to understand what is happening. Could you please explain what do you mean by setting LANG=c and LC_ALL= using export? Did you mean for me to make a system variable using export? I am sorry, im not a linux pro. I would really appreciate more help.

dennisfen, 2015-12-18:
@bremenpl, @playya's advice on setting `LANG` and `LC_ALL` variables refers to your comment about translation of `make` output from Polish. Setting these sysvars according to his advice will prevent `make` from generating localized output so you won't need to translate it.

merge, 2016-11-15:
@bremenpl do you still have compilation problems under Debian? If so, please re-post your output in english. thanks

bremenpl, 2016-11-15:
Hey, thanks for your answer. To be fair with you i dropped this topic long time a go. I cant remember either I succeed or not.

merge, 2016-11-16:
thanks for replying anyways. Closed.

closed 2016-11-16 by merge