libts/tslib #37: Problem compile with uclinux toolchain

hujiafu, 2016-02-13:
export CC=arm-uclinuxeabi-gcc
export CPP=arm-uclinuxeabi-cpp
export AS=arm-uclinuxeabi-as
export AR=arm-uclinuxeabi-ar
export LD=arm-uclinuxeabi-ld

export CFLAGS="-I. -I${TOOLCHAIN}/lib/usr/include"
export LDFLAGS="-I. -I${TOOLCHAIN}/lib/usr/lib"
export LIBS="-lz -lm"

./configure --build=i686-linux --host=arm-linux prefix=${INSTALL} --enable-static=yes --enable-shared=no

make

=================================
ts_close.c:17: fatal error: dlfcn.h: No such file or directory

because the uclinux toolchain not suport -ldl , how can I compile the tslib for uclinux?

merge, 2016-11-23:
-ldl was explicitely added in Makefile.am, but is now removed. Could you test the current master's state of tslib once again? thanks

merge, 2017-02-01:
Due to it's age, it possibly being fixed anyways and to draw attention to it one more time, I'll go ahead and close this. If you have problems with the current master, I'm happy with a new issue.

closed 2017-02-01 by merge