libts/tslib #38: Drop in pressure results in mis-clicks (jitter).

EricZaluzec, 2016-02-17:
Greetings,
Using ts_test, I have observed that a drop in 'touch' pressure when drawing a straight line will create a drastic jitter across the screen in the (x) coordinate direction. This jitter can have unwanted effects if, for example, a user is dragging a 'GUI scroll-bar' on the lcd display. 

Has this ever been observed by others? Not sure if it is hardware related. Is there anyway to compensate or remove the jitter during a change or drop in pressure?

Example: drawling line.
Notice the 'X' value increase when 'Pressure' decrease.
```
1455655154.272601:    161    181    254
1455655154.288827:    159    187    256
1455655154.305037:    158    194    258
1455655154.321250:    591    297    112
1455655154.337468:    153    214    261
1455655154.353670:    151    216    262
1455655154.369877:    151    220    263
1455655154.386081:    150    223    265
1455655154.402293:    150    228    266
1455655154.418513:    149    233    267
1455655154.434731:    148    238    269
1455655154.450937:    147    243    269
1455655154.467156:    147    248    271
1455655154.483377:    146    254    272
1455655154.499594:    145    259    272
1455655154.515814:    447    351    151
1455655154.532091:    140    277    275
1455655154.548332:    139    281    277
```

Any help, feedback, or suggestions would be greatly appreciated.

kergoth, 2016-02-17:
I haven't seen this particular issue personally. Presumably you could use a dejitter plugin to catch it as an outlier and throw it out without knowledge of the relationship to pressure, but it does sound like a peculiarity of the hardware that might be best compensated for in the driver. That said, I don't deal with touchscreens all that much anymore, so possibly someone else might know better.

EricZaluzec, 2016-02-26:
Closing issue. Thanks for the input on this.
Problem was resolved. 
I was using the TI-am335x hardware and the ti_am335x_tsc.c file had a few problems found in the linux v3.12 kernel. The ti-tsc driver was the ultimate problem. Upon patching ti_am335x_tsc and having my X11 window system pick up the variance and dejitter filters from tslib, I now see a cleaner touchscreen, especially when apply low pressure to the display.
Cheers!

closed 2016-02-26 by EricZaluzec