libts/tslib #39: Zytronic BTC_ZXY100-U-OFF-32-A not working with tslib

RazZziel, 2016-03-02:
I'm trying to make the following Zytronic BTC_ZXY100-U-OFF-32-A USB touchscreen with Qt5/EGLFS on a Raspberry Pi 2 with Raspian:
```
Bus 001 Device 035: ID 14c8:0005 Zytronic 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x14c8 Zytronic
  idProduct          0x0005 
  bcdDevice            1.00
  iManufacturer           1 Zytronic Displays Limited
  iProduct                2 Zytronic Touchscreen Controller
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     219
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               8
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```
which is detected as:
```
/dev/input/event0
/dev/input/mouse0
/dev/input/by-path/platform-3f980000.usb-usb-0:1.2:1.0-event
/dev/input/by-id/usb-Zytronic_Displays_Limited_Zytronic_Touchscreen_Controller-event-if00
```

`ts_calibrate` doesn't recognice touchs at all, and neither does `ts_print`. When I use `module_raw ucb1x00` instead of `module_raw input`, `ts_print` starts recognizing activity when I slide my finger on the touchscreen, but `ts_calibrate` still refuses to work.

If it helps, this is the output of evtest when touching, sliding and raising my finger: https://clbin.com/dGBLI. It looks pretty standard to me:

* type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
* type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 3623
* type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1298
* type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0

kergoth, 2016-03-02:
Try using ts_print_raw, which prints what's coming from the device without the tslib filters and without calibration being applied. That's what ts_calibrate is using to do the calibration. If that doesn't get events, then calibration won't work. If ts_print_raw works but ts_print doesn't, I'd try disabling the tslib filters to make sure one of them isn't causing problems for this device.

RazZziel, 2016-03-03:
With `module_raw input`, ts_print_raw` doesn't print anything (and neither does `ts_print`), but with `module_raw ucb1x00`, `TSLIB_TSDEVICE=/dev/input/event0 ts_print_raw` does get events, and so does `ts_print`.

This is what `ts_print_raw` shows with `ucb1x00` after a quick touch:
```
3735555.000054:  22232  51233   4406
3473411.002078:  22232  51233   4406
3538947.001979:  22232  51233   4406
21626881.000001:  22232  51233   4406
3.002078:  22232  51233   4406
65539.001979:  22232  51233   4406
0.000000:  22232  51233   4406
3735555.-00001:  22232   1656   4406
21626881.000000:  22232   1656   4406
0.000000:  22232   1656   4406
```
and after a slide
```
3735555.000055:  22232  45507   4455
3473411.001804:  22232  45507   4455
3538947.001465:  22232  45507   4455
21626881.000001:  22232  45507   4455
3.001804:  22232  45507   4455
65539.001465:  22232  45507   4455
0.000000:  22232  45507   4455
3473411.001826:  22232  34457   4455
3538947.001468:  22232  34457   4455
3.001826:  22232  34457   4455
65539.001468:  22232  34457   4455
0.000000:  22232  34457   4455
3473411.001833:  22232  42476   4455
3538947.001469:  22232  42476   4455
3.001833:  22232  42476   4455
65539.001469:  22232  42476   4455
0.000000:  22232  42476   4455
3473411.001843:  22232  50450   4455
3538947.001471:  22232  50450   4455                                                                                                                              
(...)
```

The output of `ts_print` is very similar. Still, `ts_calibrate` doesn't react, even with `ucb1x00` and with a working `ts_print`.

merge, 2016-11-23:
hm. I have no explanation here. I would only ask you to test once again with the current master's version of tslib. That would be great.

merge, 2017-02-04:
please open a new issue (or reopen this one), if the problem persists using tslib 1.4 and the input module. thanks!

closed 2017-02-04 by merge