libts/tslib #40: autogen.sh issue

neostar63, 2016-04-05:
executing the autogen.sh script results in failure:
`autoreconf: 'configure.ac' or 'configure.in' is required`

It appears that the syntax of the autoreconf command is off, should 'pwd' really be "$(pwd)" or some derivative of that?

kergoth, 2016-04-05:
``pwd`` and `$(pwd)` are equivalent in every shell. It sounds like you just didn't run it from inside the tslib source tree.

neostar63, 2016-04-05:
I cloned tslib and navigated to that folder and ran autogen.sh inside of the folder. Are you saying I should be executing the script from inside the src folder instead of the root for the project?

neostar63, 2016-04-05:
looking at it closer it looks like it should have been `pwd`/m4 instead of single quotes.

neostar63, 2016-04-05:
I apologize for the confusion, this must have been something on my end.

closed 2016-04-05 by neostar63