libts/tslib #43: tslib 1.1 with egalax 7 inch touchscreen

farfons, 2016-10-23:
hi, i've tried the tslib from git with a 7 inch egalax touchscreen (oeef:0001) in the Daphile distro. I can't calibrate nor with the tslib plugin (the system becomes unresponsive and i can only force the shutdown) nor with the "input" option in the ts.conf (appears the message `selected device is not a touchscreen (must support ABS and KEY event types`) can i aaply any patch or set any other variable to solve the issue? thanks

merge, 2016-11-09:
could you post an output of evtest, selecting the original device's input device?

merge, 2016-11-28:
You should be able to use the "input" option in ts.conf now, with version 1.3-rc1

merge, 2017-03-09:
well. in case you have problems using the latest version, please open a new issue. thanks!

closed 2017-03-09 by merge