libts/tslib #44: Odd y-axis reversal in program, but not in ts_test

ggalt, 2016-11-06:
I'm facing an odd situation.  I have an old Elo 1024X768 touchscreen (resistive touch) that I am using with[piCorePlayer](https://sites.google.com/site/picoreplayer/home) and a Raspberry Pi v3.  PiCorePlayer is based on [Tinycore Linux](http://tinycorelinux.net/).

When I run ts_calibrate, I get a pointercal file of:
`24774 189 -19516112 -105 -19047 64158116 65536 1024 768`

This works beautifully with ts_test and accurately tracks my touch input.  Unfortunately, with piCorePlayer, the Y-axis is inverted.  In an attempt to counter the inverted Y-axis in the piCorePlayer program, I intentionally Y-axis-inverted my responses to ts_calibrate, resulting in a pointercal file of:
`24943 157 -19874928 120 19281 -14267136 65536 1024 768`

As expected, the Y-axis of ts_test was inverted.  However the Y-axis of the piCorePlayer program continued to be inverted as well.

The program I'm using in the piCorePlayer distribution is called "Jivelite" and uses the SDL library to draw to the screen.  Is there any known interaction where SDL could be interfering with or somehow altering the information from tslib?  Are there different means of reading the output from tslib such that ts_test could be using one and piCorePlayer using another?  

I've got an inquiry in over at the piCorePlayer site too, but I thought that there might be deeper knowledge here of how tslib and SDL interact.

Thanks.

merge, 2016-11-09:
there aren't really different means of reading ts_sampes of ts_read(). One input sync-frame produces a ts_sample and that's it. Your issue sounds like your applications doesn't use ts_read() at all. Are you sure you don't use the input system?

I don't know if any tslib user manipulates ts_samples, but if one does, it doesn't sound like the right thing to do.

In case you actually don't use ts_read(): In my opinion it shouldn't even be necessary in order to use tslib. That's why I'm planning to add tslib/tools/ts_uinput that translates tslib's samples back to a normal input event device. The bad news is, it may still take a few weeks for it to come in here. It should, after all, be a reference for doing that and make ts_read() unnecessary. Applications shouldn't need to have tslib support built in.

merge, 2016-11-17:
As I see it, the SDL library simply doesn't use ts_read(). I'll close this now as this then has nothing to do with tslib. If I'm wrong, please open a new issue with detailed information on how ts_read() is used when it's not working for you.

Although with tslib 1.3, there should be coming a translator daemon one can use to have a normal input device again. This would enable SDL to use tslib without changes.

closed 2016-11-17 by merge