libts/tslib #63: Error on cross compilation on input_raw

catslab, 2017-01-30:
Hi,

I'm trying to cross compile le tslib on OSX with linaro toolchain.
I checked out master and release/1.3 and ran into the same behavior.
Note: I have an older version 1.1.0 that I can compile fine exactly the same way.

I launch config/compile with: 
```
export CC=/usr/local/linaro/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
export CXX=/usr/local/linaro/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++
export CONFIG_SITE=arm-linux.autogen
./autogen.sh
./configure --host=arm-linux-gnueabihf --prefix /usr/local/opt/qt5/tslib -enable-static=yes -enable-shared=yes
make
make install
```

Error:
```
Making all in plugins
depbase=`echo input-raw.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
	/bin/sh ../libtool  --tag=CC   --mode=compile /usr/local/linaro/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I..  -I../src  -DTS_POINTERCAL=\"/usr/local/opt/qt5/tslib/etc/pointercal\"  -DTSLIB_INTERNAL -fvisibility=hidden -DGCC_HASCLASSVISIBILITY -O2 -Wall -W -MT input-raw.lo -MD -MP -MF $depbase.Tpo -c -o input-raw.lo input-raw.c &&\
	mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  /usr/local/linaro/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc -DHAVE_CONFIG_H -I. -I.. -I../src -DTS_POINTERCAL=\"/usr/local/opt/qt5/tslib/etc/pointercal\" -DTSLIB_INTERNAL -fvisibility=hidden -DGCC_HASCLASSVISIBILITY -O2 -Wall -W -MT input-raw.lo -MD -MP -MF .deps/input-raw.Tpo -c input-raw.c  -fPIC -DPIC -o .libs/input-raw.o
input-raw.c: In function ‘check_fd’:
input-raw.c:49:19: error: ‘SYN_MAX’ undeclared (first use in this function)
 # define SYN_CNT (SYN_MAX+1)
                   ^
input-raw.c:87:30: note: in definition of macro ‘DIV_ROUND_UP’
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
input-raw.c:116:14: note: in expansion of macro ‘BITS_TO_LONGS’
  long synbit[BITS_TO_LONGS(SYN_CNT)];
              ^
input-raw.c:116:28: note: in expansion of macro ‘SYN_CNT’
  long synbit[BITS_TO_LONGS(SYN_CNT)];
                            ^
input-raw.c:49:19: note: each undeclared identifier is reported only once for each function it appears in
 # define SYN_CNT (SYN_MAX+1)
                   ^
input-raw.c:87:30: note: in definition of macro ‘DIV_ROUND_UP’
 #define DIV_ROUND_UP(n,d) (((n) + (d) - 1) / (d))
                              ^
input-raw.c:116:14: note: in expansion of macro ‘BITS_TO_LONGS’
  long synbit[BITS_TO_LONGS(SYN_CNT)];
              ^
input-raw.c:116:28: note: in expansion of macro ‘SYN_CNT’
  long synbit[BITS_TO_LONGS(SYN_CNT)];
                            ^
input-raw.c:116:7: warning: unused variable ‘synbit’ [-Wunused-variable]
  long synbit[BITS_TO_LONGS(SYN_CNT)];
       ^
input-raw.c: In function ‘ts_input_read_mt’:
input-raw.c:500:11: error: ‘ABS_MT_TOOL_X’ undeclared (first use in this function)
      case ABS_MT_TOOL_X:
           ^
input-raw.c:505:11: error: ‘ABS_MT_TOOL_Y’ undeclared (first use in this function)
      case ABS_MT_TOOL_Y:
           ^
make[2]: *** [input-raw.lo] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

merge, 2017-01-31:
Well yes. Since tslib 1.3 the input plugin basically only supports recent Linux kernels, because we support multitouch. For now, you can use tslib without it (`./configure --disable-input` and probably you'll need `--disable-tools` too) or use tslib 1.2.

Ideally we should provide single touch if (like in your case) SYN_CNT or some ABS_MT_* is not defined. I'd be happy to take a pull request doing that.

merge, 2017-01-31:
Would you be able to pull the latest changes in master and build it again for me? Thanks!

merge, 2017-02-02:
Well, this particular build error is fixed and version 1.4 is around the corner, so I'll close this. Thank you very much for reporting this! Please don't hesitate to report bugs in the future.

thanks

smartphoneworld, 2017-03-29:
Hi
I can see this issue on latest master. I also tried pulling 1.4 and can see the issue as well. Any idea?
input-raw.c: In function 'ts_input_read_mt':
input-raw.c:520:11: error: 'ABS_MT_TOOL_X' undeclared (first use in this function)
input-raw.c:520:11: note: each undeclared identifier is reported only once for each function it appears in
input-raw.c:525:11: error: 'ABS_MT_TOOL_Y' undeclared (first use in this function)

Thanks

smartphoneworld, 2017-03-30:
I was trying to fix the issue.. I found my input.h header file dose not include those two macros. A fast fix for that was adding those two missing macros to my input.h (I got them from newer versions of input.h) I add them like this:
cat >> /usr/include/linux/input.h << __EOF__
#undef ABS_MAX
#undef ABS_CNT
#define ABS_MT_TOOL_X           0x3c    /* Center X tool position */
#define ABS_MT_TOOL_Y           0x3d    /* Center Y tool position */
#define ABS_MAX                 0x3f
#define ABS_CNT                 (ABS_MAX+1)
__EOF__

Also those macros were moved on newer kernel headers to input-event-codes.h
Would you please confirm I am on the right path. 

Thanks.

merge, 2017-03-30:
well. the move to input-event-codes.h doesn't matter for users. input.h pulls this is of course. Editing your system headers is not really the right path here. Fixing tslib is, so reporting the bug is the right path :)

Can you confirm the current master is affected? Please make sure you have pulled the latest state from this week or today, cleaned your repository, and please paste the complete compile errors you get in here.

thanks!

smartphoneworld, 2017-03-30:
Sure, I tried with the latest master as well as with 1.4.. Both were pulled today an hour or two before writing here, and both had the same complain but in different files. Complain on master was following:
make[2]: Entering directory `/root/tslib/tools'
  CC     ts_uinput.o
ts_uinput.c: In function 'send_touch_events':
ts_uinput.c:216:23: error: 'ABS_MT_TOOL_X' undeclared (first use in this function)
ts_uinput.c:216:23: note: each undeclared identifier is reported only once for each function it appears in
ts_uinput.c:222:23: error: 'ABS_MT_TOOL_Y' undeclared (first use in this function)
make[2]: *** [ts_uinput.o] Error 1
make[2]: Leaving directory `/root/tslib/tools'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/root/tslib'

version 1.4 pulled today:
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I.. -I../src -DTS_POINTERCAL=\"/usr/local/etc/pointercal\" -DTSLIB_INTERNAL -fvisibility=hidden -DGCC_HASCLASSVISIBILITY -O2 -Wall -W -MT input-raw.lo -MD -MP -MF .deps/input-raw.Tpo -c input-raw.c  -fPIC -DPIC -o .libs/input-raw.o
input-raw.c: In function 'ts_input_read_mt':
input-raw.c:520:11: error: 'ABS_MT_TOOL_X' undeclared (first use in this function)
input-raw.c:520:11: note: each undeclared identifier is reported only once for each function it appears in
input-raw.c:525:11: error: 'ABS_MT_TOOL_Y' undeclared (first use in this function)
make[2]: *** [input-raw.lo] Error 1
make[2]: Leaving directory `/root/tslib/plugins'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/root/tslib'

Thank you for your advise.

merge, 2017-03-30:
1.4 is old. Please use 1.8 (or master, only for testing) and do `./configure --disable-tools`. What kernel version are you using. This is known; ts_uinput is only supported on reasonably new kernels.

merge, 2017-03-30:
Specifically Linux 3.5 and older are required to do `./configure --disable-tools`. This still is a bug, if you will, so I'll keep this issue open and see how we can automate this.

merge, 2017-03-30:
Ok. I decided to push back this issue by including the input definitions in ts_uinput too. This has nothing to do with input_raw.

If you have issues with ts_uinput using the current master, please open a new issue.

Thanks for taking the time and reporting your issues!

closed 2017-03-30 by merge

smartphoneworld, 2017-03-30:
./configure --disable-tools worked.. I appreciate your prompt support... However, I am on kernel 3.8.. here is my uname -a:
Linux beaglebone 3.8.13-bone71 #1 SMP Tue Mar 17 18:07:44 UTC 2015 armv7l GNU/Linux

Thanks. :)