libts/tslib #64: start documenting in the wiki

merge, 2017-02-02:
we would have enough material from the man pages and the README, we could present more intelligently
* API
* library ABI, versioning
* examples
* state of the symbols
* better filter module descriptions
* related libraries (input stack)
* 3rd party programs using tslib's API
* tutorials
* how to write a filter module
* how to report bugs
* how to contribute
* ...

AudioMorphology, 2017-02-03:
I don't know if it is any help, but I documented my specific tslib install and setup process in the wiki for my own project:

https://github.com/AudioMorphology/Europi/wiki/Install-and-configure-TSLIB

I am using tslib to handle touchscreen input on a 320x240 pixel TFT on a Raspberry Pi, so my instructions are very specific and probably not general enough?

merge, 2017-02-03:
That's a nice example tutorial for users. Thanks a lot!

And please feel free to report any problems

merge, 2017-02-03:
@AudioMorphology note that by using the variance module you lose multitouch. In case your device is a multitouch device, I recommend playing with the median module instead.

merge, 2017-02-07:
The wiki is active: https://github.com/kergoth/tslib/wiki and let's see what happens. Let's keep it simple with single self-contained pages only.

merge, 2017-02-09:
well. the wiki ist started. TODOs indicate missing things on the pages. No need for this issue anymore.

closed 2017-02-09 by merge