libts/tslib #65: Missing x64 support

pfink-christ, 2017-02-07:
Hi,
I encountered multiple issues, while trying to get ts_calibrate & co to work properly on x64 machines (Ubuntu/Debian).

There are multiple places where hardcoded lengths cause the applications to crash with segfaults.
[For example:](https://github.com/kergoth/tslib/blob/master/tests/fbutils.c#L155)    `line_addr = malloc (sizeof (__u32) * var.yres_virtual);`

> *** Error in `ts_calibrate': malloc(): memory corruption: 0x0000000001ab1130 ***
> Segmentation fault.

or

> *** Error in `ts_calibrate': double free or corruption (!prev): 0x000000000068d040 ***
> Segmentation fault.

I already fixed the parts that were important to me.
Is anybody interested in x64 support? Then I will try to get everything running and send a pull request when ready.

merge, 2017-02-08:
I'd happily take a pull request fixing 64 bit. Thanks for bringing this up!

pfink-christ, 2017-02-09:
Ok, here it is https://github.com/pfink-christ/tslib/commit/c4c59790f3191c1a233fc1a61f8fedad85de1aeb
It really was that single line which caused the problems.

I also fixed some memory leaks that I saw while debugging with valgrind and fixed the buttons which were broken for me.

PR #68 (2017-02-09): Add x64 compatibility, fix some memory leaks, fix buttons
Fix #65 and improve testutils by fixing memory leaks and the button drawing
commit 93137ba4 (2017-02-09): Merge pull request #68 from pfink-christ/master

Add x64 compatibility, fix some memory leaks, fix buttons

Fixes #65
closed 2017-02-09 by merge