libts/tslib #72: ts_calibrate and ts_test_mt: add rotation option

merge, 2017-02-20:
we could add rotating 90, 180 and 270 degrees. The raw framebuffer doesn't do that for us.

radianresearch, 2017-05-08:
I have an immediate need for this. Has any work along these lines been started?

merge, 2017-05-08:
Github is acting weird, I saw somebody asking "has work on this started?" but I don't see it in the issue here. whatever.

I haven't started working on this. But it's mainly a matter of defining what to do. That's where you can help too. I've seen workarounds to this somewhere, but I'd like to do it properly.

* Basically we should implement the `ts_option()` API
* I guess we should offer the 3 obvious rotation options 90 degrees CW and CCW and 180 degrees.
* How should the rotate integer work? What's intuitive and others use as well? X.org?
* Implementing the framebuffer display in the graphical applications would we quite easy.
* What exactly, if anything, has to be done to the x/y values, and why?

merge, 2017-05-08:
Just to make `ts_option()` clear,

    enum ts_param {
            TS_SCREEN_RES = 0,              /* 2 integer args, x and y */  
            TS_SCREEN_ROT                   /* 1 integer arg, 1 = rotate */
    };

is defined for users, and one would call `ts_option(tsdev, TS_SCREEN_ROT, <rotation_value>)` and the rotation_value is to be defined and used.

radianresearch, 2017-05-08:
@merge That was me. I decided that since the issue was on hold, the answer was probably "nothing done yet". I ended up kludging how fbutils.c draws characters/strings, so on our display they would appear right-side up. We don't need 90 or 270 rotation, hardcoding a 180 rotation is working for us. I can supply a patch if anyone wants it.

merge, 2017-05-08:
ok. of course I'd be happy to see your changes, even if it sounds like a workaround too :)

merge, 2017-05-09:
I guess we'll implement said rotation value like the framebuffer console does is:
```

4. fbcon=rotate:<n>

        This option changes the orientation angle of the console display. The
        value 'n' accepts the following:

	      0 - normal orientation (0 degree)
	      1 - clockwise orientation (90 degrees)
	      2 - upside down orientation (180 degrees)
	      3 - counterclockwise orientation (270 degrees)

	The angle can be changed anytime afterwards by 'echoing' the same
	numbers to any one of the 2 attributes found in
	 /sys/class/graphics/fbcon

		rotate     - rotate the display of the active console
		rotate_all - rotate the display of all consoles

```

merge, 2017-05-16:
I tend to leave ts_option() and the library completely untouched and only implement framebuffer roation for ts_calibrate / ts_test / ts_test_mt. (only library users)

* graphical environments have their own rotating mechanism and rotate input AND screen
* so we don't rotate input and ts_calibrate only __displays__ rotated, to improve the UX.

merge, 2017-05-21:
Implemented for ts_test_mt and ts_calibrate. As lined out here, only framebuffer output is rotatet. Input sample rotation has to be done in filters (for example re-calibrating).

It's a seperate thing. And if we implement ts_option() to switch on-the-fly has yet to be seen.

closed 2017-05-21 by merge