libts/tslib #75: build on Mac OS X

merge, 2017-03-09:
* build libts and filters under OSX and fix possible issues in the process.
* document which module_raw builds too (wiki)
* which of the test programs build and work?

merge, 2017-04-12:
unknown platforms build just fine now. easy to do when having git, autoconf and automake.

DrankoLQ, 2017-06-08:
I did execute the commands and got an error on the configure phase:

`
...
checking that generated files are newer than configure... done
configure: error: conditional "ENABLE_UCB1X00_MODULE" was never defined.
Usually this means the macro was only invoked conditionally.
`

macOS Sierra 10.12.4

merge, 2017-06-08:
did you use the suggested version 1.8? if so, could you retry with 1.11?

DrankoLQ, 2017-06-08:
Yes, 1.8 failed but 1.11 works perfect! Thanks :)

merge, 2017-06-09:
it would be interesting, what modules build. If you want to, use --enable-corgi and so on and collect a list of modules that work :)