libts/tslib #76: Multi-touch is not functional in non-blocking mode

ClayMontgomery, 2017-03-09:
I upgraded my app (Lightwing) from tslib 1.0 to 1.5 to use the new multi-touch features. It is working in blocking mode, but when tslib is initialized for non-blocking operation (ts_setup(NULL, 1) or ts_open(NULL, 1)), this causes two problems. The first problem is ts_read_mt() always returns -1, unless the variance filter is used (which of course invalidates the multi-touch data). The second problem is the annoying message "tslib: error reading input event: Resource temporarily unavailable" is emitted on every call to ts_read_mt(), regardless of the filters used. That issue can be resolved by simply removing the single perror() call in input-raw.c, but I could use some help to resolve the first problem.

I would think the non-blocking mode is the more common use case, but all of the tslib tests and example codes use only the blocking mode. That is probably why this problem has not been reported yet.

I am using tslib 1.5 on Linux 3.14.28 built from Yocto 1.8.2 for the NXP i.MX6.

Regards, Clay

merge, 2017-03-09:
thanks for bringing this up! We obviously never use nonblocking, but Qt for example does too and wouldn't be able to switch to ts_read_mt.

we should for example add a --non-bocking option to our test programs.

I'll most likely look at it more closely next week. Thanks for this report! Of course I'd be happy to discuss a pull request too.

merge, 2017-03-10:
The annoying error message is a bug definitely! Thanks!
The variance filter doesn't propagate errno, which is a problem too, but we don't use it with multitouch.

You get -EAGAIN when using it non-blocking, which can be ok. I'm not yet sure whether we a have a real bug here. Could you:

- test if `ts_read()` is causing the same problem as `ts_read_mt()` for you?
- post a code-snipped, how you use it.

Thanks for your help.

merge, 2017-03-10:
The error message is removed in master. Could you check if the following change fixes your problem:

```
diff --git a/plugins/input-raw.c b/plugins/input-raw.c
index 9d26bd7..0a84225 100644
--- a/plugins/input-raw.c
+++ b/plugins/input-raw.c
@@ -456,7 +456,10 @@ static int ts_input_read_mt(struct tslib_module_info *inf,
                        rd = read(ts->fd,
                                  i->ev,
                                  sizeof(struct input_event) * NUM_EVENTS_READ);
-                       if (rd < (int) sizeof(struct input_event)) {
+                       if (rd == -1) {
+                               total = errno;
+                               break;
+                       } else if (rd < (int) sizeof(struct input_event)) {
                                total = -1;
                                break;
                        }

```

merge, 2017-03-10:
By the way, this is how Qt uses `ts_read()` (simplified) opened non-blocking:

```
static bool get_sample(struct tsdev *dev, struct ts_sample *sample){
        return (ts_read(dev, sample, 1) == 1);
}
```

which should be possible with `ts_read_mt()` too...

ClayMontgomery, 2017-03-10:
I have confirmed that ts_read() is not causing the same problem as ts_read_mt().  In other words, ts_read(0 works correctly (does not always return -1) in non-blocking mode.

Here is a code snippet of how I use it:

#define MAX_TOUCH_SAMPLES	10
#define MAX_TOUCH_SLOTS		2

static struct ts_sample_mt aTouchScreenSamples[MAX_TOUCH_SAMPLES][MAX_TOUCH_SLOTS];
struct ts_sample_mt * pSample = &aTouchScreenSamples[0][0];

iSampleCount = ts_read_mt(ts, &pSample, MAX_TOUCH_SLOTS, MAX_TOUCH_SAMPLES);

merge, 2017-03-11:
ts_read_mt() now returns the -errno that read() sets. so if you can use read() non-blocking, you should be covered. Am I mistaken?

thanks

merge, 2017-03-11:
You made me fix our return value mess! Thanks again. Basically I consider this issue fixed for ts_read_mt(). I'll release 1.7 soon and you can count on consistent return values now. So, for non-blocking, simply catch -11 (or other blocking-related error codes).

I won't change ts_read() behaviour. Who knows what people are doing with it...

ClayMontgomery, 2017-03-12:
Excellent! I will test 1.7 when released.

Thanks, Clay

merge, 2017-03-13:
it is.

ClayMontgomery, 2017-03-14:
I am now testing tslib 1.7 with this ts.conf:
    module_raw input
    module pthres pmin=1
    module linear

It is somewhat improved from before, but non-blocking mode only works when only a single sample is read, such as:
    ts_read_mt(ts, pSamples, SLOTS, 1);

Any higher number of samples requested causes ts_read_mt() to usually return -1.  It seems to get worse with higher numbers of samples requested; 2 samples usually fails, 5 samples always fails.

So, when reading multiple samples, it does not seem to be possible to detect how many samples were actually read.

This problem does not occur in blocking mode.

In my experience, the ability to read multiple samples in non-blocking mode is absolutely essential to building a responsive touch-screen user interface, without having to dedicate a separate thread specifically to reading from tslib.

merge, 2017-03-15:
we badly need a test to catch all these cases. I'll fix this as time permits. thanks for testing!

merge, 2017-03-17:
The issue should be fixed in master. Except that I found that the skip filter is broken in our usecase here. I can't fix that quickly now and opened an issue. I'll do a release soon.

ClayMontgomery, 2017-03-19:
Thanks! I will test it this week.

merge, 2017-03-20:
Sorry to have you waiting. This fix is important and I want it in a release soon too. You can expect 1.8 to be released in about 24 hours.

merge, 2017-03-21:
tslib 1.8 is out. you should be able to use ts_read_mt() non-blocking and get the samples you need :) I'd be happy if you could check back here and confirm / close or complain. thanks!!

merge, 2017-03-23:
I go forth and close. Please feel free to report back anyways, or reopen in case of problems. thanks again for your help.

I put you in our THANKS file if that's ok. If you want to, give me an email address for it too.

closed 2017-03-23 by merge

ClayMontgomery, 2017-03-25:
I tested tslib release 1.8 and confirmed that the issues with multi-touch in non-blocking mode are fixed. I have also tried reading multiple samples in non-blocking mode and that works too. Thanks for the help with this!