libts/tslib #79: Segfault when reading multiple samples.

ClayMontgomery, 2017-03-10:
When the 4th parameter (MAX_TOUCH_SAMPLES) to ts_read_mt() is greater than 1, it causes a segfault. This problem occurs for me in both blocking and non-blocking modes.

I am using a static array for the samples, rather than the dynamic allocations in the examples, but this simplifies the code.

Also, I don't understand why the 2nd parameter to ts_read_mt() is typed as ts_sample_mt **, because it is not used to modify the input pointer. I think it should have one less indirection: ts_sample_mt *

Here is a code snippet of how I use it:

#define MAX_TOUCH_SAMPLES	10
#define MAX_TOUCH_SLOTS		2

static struct ts_sample_mt aTouchScreenSamples[MAX_TOUCH_SAMPLES][MAX_TOUCH_SLOTS];
struct ts_sample_mt * pSample = &aTouchScreenSamples[0][0];

iSampleCount = ts_read_mt(ts, &pSample, MAX_TOUCH_SLOTS, MAX_TOUCH_SAMPLES);

Regards, Clay

merge, 2017-03-11:
I can't reproduce this a first glance. Please post your ts.conf configuration file!

Thanks for reporting this!

merge, 2017-03-11:
ok. There's probably a more elegant way, but the following works:

```
static struct ts_sample_mt aTouchScreenSamples[SAMPLES][SLOTS];

struct ts_sample_mt (*pTouchScreenSamples)[SLOTS] = aTouchScreenSamples;
static struct ts_sample_mt *pSample[SAMPLES];
for (i = 0; i < SAMPLES; i++)
        pSample[i] = pTouchScreenSamples[i];

```
and call ts_read_mt:

`ts_read_mt(ts, pSample, SLOTS, SAMPLES);`

... but the reason for this problem is bad API design. You're right. The pointer would have been enough.

ClayMontgomery, 2017-03-12:
Thanks for this code example! This resolves the problem for me. The segfault was because I was only initializing the pointer for the first sample and not the others.

I think the added complexity of using 2 arrays should be unnecessary, but I like that at least the dynamic allocations are avoided for each call to ts_read_mt(), by using static arrays. 

Thanks, Clay

merge, 2017-03-12:
I mention it in the README too. cheers!

closed 2017-03-12 by merge