libts/tslib #82: project organisation / tslib.org / mailing list

merge, 2017-03-15:
First of all: **This github repository will stay the official place for development and releases!**
... as long as possible. there's no real reason to change in sight.

Primarily I want to have a mailing list again. But this would then be a random second place of development besides github. The I tought about putting random unofficial binary package up somewhere, and things would be a total mess :) So: there is http://www.tslib.org/ now.

tslib.org should stay one simple index.html file, pointing to everything tslib. (it need a logo though :). Furthermore, *despite* having the official development in github, I'd love to be completely independent of it. We explicitely use github's issue tracker, but that's it.

bla bla. That's my todo list:

 - [ ] register http://tslib.org
 - [ ] register https://gitlab.com/tslib/tslib (only mirrors this repo once a day)
 - [ ] upload release tarballs to gitlab too
 - [ ] remove unnecessary absolute paths that include github.com
 - [ ] have a mailing list (where?) and a link at tslib.org
 - [ ] document everything for future maintainers

ClayMontgomery, 2017-03-15:
Regarding the need for a logo, how about something like this:

[image]

Many logos like this are available here for just a few dollars:

https://www.iconfinder.com/icons/351220/finger_fingers_gesture_hand_ripple_tap_touch_wave_icon

merge, 2017-03-16:
That's awfully nice to have an example! To be honest, I have an idea. I'd like it to be basically the name "tslib", but maybe really drawn by hand, and the "b" to be a hand, similar to your image :)

I can't explain to too well, but the drawn "tslib" word could have some hints to x/y math stuff, hm. Maybe I should get a pen and post roughly what I mean. I'll do sometime soon.

merge, 2017-03-22:
by the way. That's what I'm thinking about: https://github.com/kergoth/tslib/issues/73 what would you say?

merge, 2017-03-24:
2a8b968 add the mailing list to all relevant pages

closed 2017-03-24 by merge