libts/tslib #84: Some eGalax controllers do not work correctly

neilwhelchel, 2017-03-17:
I have several eGalax touch screen controllers that all have the same USB id: 0eef:0001.
Some of them fail to work because there is no other pressure reported than 255, others work fine because when you remove your finger from the touch screen there is always a duplication of the last X/Y coordinate with a pressure of 0.

merge, 2017-03-18:
what's your ts.conf that doesn't work?

neilwhelchel, 2017-03-18:
I tried both:
module_raw input
module dejitter delta=100
module linear

And:
module_raw galax
module dejitter delta=100
module linear

Both configurations work with all of the eGalax controllers I have, however the later ones that have both serial and USB are the ones that I never see the pressure 0 from. The older ones send 255, and there is a single 0 at the time I remove my finger from the screen.

If you look in testutils.c, you see that there are 2 loops in getxy():
do {
    ...
} while (samp[0].pressure == 0);
...
do {
    ...
} while (samp[index].pressure > 0);

The first loop only exits if pressure is non zero, so the code there only gets executed on entry, which is what seems to be what the author intended.
The second loop only exits if pressure is zero, so this loop is never exited unless there is at least one event with a pressure of 0 (or lower if possible).
Since some of the controllers never send a pressure of 0, this is where the things get stuck.

It seems that the lower level driver should be sending an event with the pressure of zero, so this issue could originate from the kernel all the way down to the hardware itself, and is likely where the issue is between the two controllers. However, there probably should be a filter for tslib that generates a 0 pressure event with the same X/Y coordinates as the previous event if a new event has not arrived within a specified time. (Or maybe since this is an issue related to the eGalax controller, it could be added to the galax input module.)

merge, 2017-03-18:
From those newer ones that don't work, what would be really helpful, is a **complete** output of `evtest` selecting your device and doing a very short one-finger movement.

Anyways, when using `input_raw`, please check if ts_test_mt or ts_print_mt work better than ts_test or ts_print. They use the new ts_read_mt() call. ts_calibrate or ts_harvest use the old ts_read() call.

The driver is _not_ required to send pressure 0. Our input raw plugins generate it in case it doesn't because tslib internally relies on pressure values. So the issue most probably should get fixed in our raw plugin. I'd see clearer with evtest output of the original device. I don't really _want_ to change ts_read(), but we'll see.

neilwhelchel, 2017-03-18:
Good point about using ts_test_mt, but my point here is that ts_calibrate uses getxy() and there is no way to calibrate the screen if this fails.

Below is the output of evtest with the older (working) eGalax controller,
There is both a BTN_TOUCH value=1, and a BTN_TOUCH value=0 included in the events marking the start and end of the touch event:

evtest /dev/input/event5
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xeef product 0x1 version 0x100
Input device name: "eGalax Inc. USB TouchController"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    578
      Min        0
      Max     2047
    Event code 1 (ABS_Y)
      Value   1803
      Min        0
      Max     2047
Properties:
Testing ... (interrupt to exit)
Event: time 1489329996.884521, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1489329996.884521, type 3 (EV_ABS), code 0 (ABS_X), value 433
Event: time 1489329996.884521, type 3 (EV_ABS), code 1 (ABS_Y), value 878
Event: time 1489329996.884521, -------------- SYN_REPORT ------------
Event: time 1489329996.892510, type 3 (EV_ABS), code 0 (ABS_X), value 433
Event: time 1489329996.892510, type 3 (EV_ABS), code 1 (ABS_Y), value 878
Event: time 1489329996.892510, -------------- SYN_REPORT ------------
Event: time 1489329996.904507, type 3 (EV_ABS), code 0 (ABS_X), value 433
Event: time 1489329996.904507, type 3 (EV_ABS), code 1 (ABS_Y), value 878
Event: time 1489329996.904507, -------------- SYN_REPORT ------------
Event: time 1489329996.912506, type 3 (EV_ABS), code 0 (ABS_X), value 433
Event: time 1489329996.912506, type 3 (EV_ABS), code 1 (ABS_Y), value 878
Event: time 1489329996.912506, -------------- SYN_REPORT ------------
Event: time 1489329996.924509, type 3 (EV_ABS), code 0 (ABS_X), value 433
Event: time 1489329996.924509, type 3 (EV_ABS), code 1 (ABS_Y), value 878
Event: time 1489329996.924509, -------------- SYN_REPORT ------------
Event: time 1489329996.936509, type 3 (EV_ABS), code 0 (ABS_X), value 432
Event: time 1489329996.936509, type 3 (EV_ABS), code 1 (ABS_Y), value 876
Event: time 1489329996.936509, -------------- SYN_REPORT ------------
Event: time 1489329996.948507, type 3 (EV_ABS), code 0 (ABS_X), value 431
Event: time 1489329996.948507, type 3 (EV_ABS), code 1 (ABS_Y), value 875
Event: time 1489329996.948507, -------------- SYN_REPORT ------------
Event: time 1489329996.960509, type 3 (EV_ABS), code 0 (ABS_X), value 431
Event: time 1489329996.960509, type 3 (EV_ABS), code 1 (ABS_Y), value 874
Event: time 1489329996.960509, -------------- SYN_REPORT ------------
Event: time 1489329996.968508, type 3 (EV_ABS), code 0 (ABS_X), value 431
Event: time 1489329996.968508, type 3 (EV_ABS), code 1 (ABS_Y), value 876
Event: time 1489329996.968508, -------------- SYN_REPORT ------------
Event: time 1489329996.980510, type 3 (EV_ABS), code 0 (ABS_X), value 427
Event: time 1489329996.980510, type 3 (EV_ABS), code 1 (ABS_Y), value 881
Event: time 1489329996.980510, -------------- SYN_REPORT ------------
Event: time 1489329996.996509, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1489329996.996509, type 3 (EV_ABS), code 0 (ABS_X), value 427
Event: time 1489329996.996509, type 3 (EV_ABS), code 1 (ABS_Y), value 881
Event: time 1489329996.996509, -------------- SYN_REPORT ------------

Here is the output of evtest with the new eGalax (not working) controller, there is only BTN_TOUCH value=1, the BTN_TOUCH value=0 never comes in:

evtest /dev/input/event6
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xeef product 0x1 version 0x210
Input device name: "eGalax Inc. USB TouchController"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
[... 212 lines cut ...]
Event: time 1489331119.366208, type 3 (EV_ABS), code 59 (ABS_MT_DISTANCE), value 0
Event: time 1489331119.366208, type 3 (EV_ABS), code 0 (ABS_X), value 1784
Event: time 1489331119.366208, type 3 (EV_ABS), code 1 (ABS_Y), value 2200
Event: time 1489331119.366208, -------------- SYN_REPORT ------------
Event: time 1489331119.372208, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1489331119.372208, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1784
Event: time 1489331119.372208, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 2202
Event: time 1489331119.372208, type 3 (EV_ABS), code 59 (ABS_MT_DISTANCE), value 0
Event: time 1489331119.372208, type 3 (EV_ABS), code 0 (ABS_X), value 1784
Event: time 1489331119.372208, type 3 (EV_ABS), code 1 (ABS_Y), value 2202
Event: time 1489331119.372208, -------------- SYN_REPORT ------------
Event: time 1489331119.424211, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1489331119.424211, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1784
Event: time 1489331119.424211, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 2202
Event: time 1489331119.424211, type 3 (EV_ABS), code 59 (ABS_MT_DISTANCE), value 0
Event: time 1489331119.424211, type 3 (EV_ABS), code 0 (ABS_X), value 1784
Event: time 1489331119.424211, type 3 (EV_ABS), code 1 (ABS_Y), value 2202
Event: time 1489331119.424211, -------------- SYN_REPORT ------------
Event: time 1489331119.426210, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1489331119.426210, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 1784
Event: time 1489331119.426210, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 2202
Event: time 1489331119.426210, type 3 (EV_ABS), code 59 (ABS_MT_DISTANCE), value 1
Event: time 1489331119.426210, type 3 (EV_ABS), code 0 (ABS_X), value 1784
Event: time 1489331119.426210, type 3 (EV_ABS), code 1 (ABS_Y), value 2202
Event: time 1489331119.426210, -------------- SYN_REPORT ------------

merge, 2017-03-18:
thanks a lot. Your device seems to use distance 1 as the only code for pen-up. I've not seen this before but I guess we can add that to our raw plugins. I probably can fix it during next week.

I have to evaluate and test a bit to see if I put it in 1.8.

thanks again

merge, 2017-03-18:
Since is broken behaviour, I'll take the time to look at usbhid before working around it in tslib. But we'll do it, one way or the other.

neilwhelchel, 2017-03-19:
If you are going to look at the driver, I am using Linux 3.10.104+ SMP PREEMPT aarch64.

merge, 2017-03-19:
Given that this is really the same device ID, behaving that different, I'll fix this broken hardware in tslib. The Linux drivers have quirks, of course, but use device IDs to differentiate.

There is a fix in master now, in the galax raw module only. That's what I can safely do for 1.8, which I want to release very soon. Can you test the current master branch of tslib, using `module_raw galax` ? That'd be great!!

(This way you won't get multitouch of course. My longer term plan is to discontinue the galax raw module development and merge it with the normal input one.)

neilwhelchel, 2017-03-19:
I tested the changes you made, and it seems that using ABS_MT_DISTANCE works just fine. I made it a point to work with it long enough to verify that ABS_MT_DISTANCE=1 is a reliable method of determining a release, and so far so good. 
Thank you for the patch!

merge, 2017-03-19:
good. thank you for the great help. I'll add you to the THANKS file if that is ok, and you could give an email adress for it too, if you want.

I'll leave closing the issue up to you for a while.

neilwhelchel, 2017-03-20:
Yes, you can add me (I appreciate that), my email is: Neil Whelchel <neil@goautomation.biz>
I will test more over the next couple of days, and if all looks good by the time we solve the graphics issue, I will close this one.

merge, 2017-03-22:
Since you use git anyways, could you test using `module_raw input` with the latest addition to master with your devices? If you want to help, test it with both the old and new ones.

And with the new ones, please also try ts_test_mt using 1 to 8 touch contacts simultaneously.

merge, 2017-03-26:
Despite I'd still like feedback on using module_raw input here, I want to tidy up and go ahead and close this.

Feel free to reply anyways. thanks a lot.

closed 2017-03-26 by merge