libts/tslib #85: Graphics issues with ts_calibrate and friends.

neilwhelchel, 2017-03-19:
With ts_calibrate, the target leaves behind pixels when it moves. The movement is animated in a few steps between each position, and at each of these steps the target remains visible. With ts_test, in the mode to drag the target around, it leaves a trail of targets instead of moving it.
There seems to be a problem with the way that fbutils-linux deals with pixel XOR operations not cleaning up pixels (properly) the second time put_cross() is called.
My configuration is 32 bpp, 8 bits each for red, green, blue, and transparency. I have not verified that the data offsets read from the kernel via ioctl are correct. The platform is aarch64.

merge, 2017-03-19:
Thanks for reporting this. I've never really touched the framebuffer code and I can't reproduce this either, so I'll get tricky for me look into it.

I'm very happy to take a pull request and test it on my systems in case you can experiment a bit on your own, fixing it for you.

merge, 2017-03-19:
Ok. The first thing I would try, is the following change:
```

--- a/tests/fbutils-linux.c
+++ b/tests/fbutils-linux.c
@@ -145,8 +145,8 @@ int open_framebuffer(void)
        memset(fbuffer,0,fix.smem_len);
 
        bytes_per_pixel = (var.bits_per_pixel + 7) / 8;
-       transp_mask = ((1 << var.transp.length) - 1) <<
-               var.transp.offset; /* transp.length unlikely > 32 */
+       transp_mask = 0;
+
        line_addr = malloc (sizeof (*line_addr) * var.yres_virtual);
        addr = 0;
        for (y = 0; y < var.yres_virtual; y++, addr += fix.line_length)

```

This was added more recently and really could be wrong in your case. It's a bit of a wild guess, but I keep you busy at least :)

neilwhelchel, 2017-03-20:
I will look into this to attempt to get an understanding of what is going on. I will report back soon.

neilwhelchel, 2017-03-20:
Ok, here is what I found. The long datatype on aarch64 is actually 64 bits, not 32 as the code was expecting. According to the C standard, the length of short, int, long and long long are not specific sizes but they are relative to each other. (POSIX defines int as 8 bits.) The framebuffer code relies on them being specific sizes, and since on aarch64, long is 64 bits instead of the 32 bits expected by the code, the xor function was working on two pixels, not one, so every other pixel was wrong after the second xor pass.
The patch(es) below addresses this problem, and includes some minor formatting cleanups, and it also addresses the issue of clearing the screen after the program finishes. (It is really annoying when the program finishes and leaves things behind.)
I would suggest looking around the rest of the code for issues like this. Also I suggest using the (standard) uintXX_t from stdint.h instead of the system specific __uXX. (My understanding is that the __uXX data types were used in the kernel because they pre-date the C99 stdint.h data types. Also, the __uXX data types are intended for use in kernel space, and it is discouraged to use kernel data types in user space programs except when required as in fb_var_screeninfo and similar.)
I included two patches below, use the one that you think is most correct.

This patch continues to use the existing data types where possible:

--- a/tests/fbutils-linux.c
+++ b/tests/fbutils-linux.c
@@ -29,12 +29,12 @@
 #include "fbutils.h"
 
 union multiptr {
-       unsigned char *p8;
-       unsigned short *p16;
-       unsigned long *p32;
+       __u8 *p8;
+       __u16 *p16;
+       __u32 *p32;
 };
 
-static int con_fd, fb_fd, last_vt = -1;
+static int con_fd, last_vt = -1;
 static struct fb_fix_screeninfo fix;
 static struct fb_var_screeninfo var;
 static unsigned char *fbuffer;
@@ -142,7 +142,7 @@ int open_framebuffer(void)
                close(fb_fd);
                return -1;
        }
-       memset(fbuffer,0,fix.smem_len);
+       memset(fbuffer, 0, fix.smem_len);
 
        bytes_per_pixel = (var.bits_per_pixel + 7) / 8;
        transp_mask = ((1 << var.transp.length) - 1) <<
@@ -157,6 +157,7 @@ int open_framebuffer(void)
 
 void close_framebuffer(void)
 {
+       memset(fbuffer, 0, fix.smem_len);
        munmap(fbuffer, fix.smem_len);
        close(fb_fd);

This patch uses (more) user space data types:

--- a/tests/fbutils-linux.c
+++ b/tests/fbutils-linux.c
@@ -20,6 +20,7 @@
 #include <sys/mman.h>
 #include <sys/time.h>
 #include <fcntl.h>
+#include <stdint.h>
 
 #include <linux/vt.h>
 #include <linux/kd.h>
@@ -29,21 +30,21 @@
 #include "fbutils.h"
 
 union multiptr {
-       unsigned char *p8;
-       unsigned short *p16;
-       unsigned long *p32;
+       uint8_t *p8;
+       uint16_t *p16;
+       uint32_t *p32;
 };
 
-static int con_fd, fb_fd, last_vt = -1;
+static int con_fd, last_vt = -1;
 static struct fb_fix_screeninfo fix;
 static struct fb_var_screeninfo var;
 static unsigned char *fbuffer;
 static unsigned char **line_addr;
 static int fb_fd=0;
 static int bytes_per_pixel;
-static int transp_mask;
+static uint32_t transp_mask;
 static unsigned colormap [256];
-__u32 xres, yres;
+uint32_t xres, yres;
 
 static char *defaultfbdevice = "/dev/fb0";
 static char *defaultconsoledevice = "/dev/tty";
@@ -142,7 +143,7 @@ int open_framebuffer(void)
                close(fb_fd);
                return -1;
        }
-       memset(fbuffer,0,fix.smem_len);
+       memset(fbuffer, 0, fix.smem_len);
 
        bytes_per_pixel = (var.bits_per_pixel + 7) / 8;
        transp_mask = ((1 << var.transp.length) - 1) <<
@@ -157,6 +158,7 @@ int open_framebuffer(void)
 
 void close_framebuffer(void)
 {
+       memset(fbuffer, 0, fix.smem_len);
        munmap(fbuffer, fix.smem_len);
        close(fb_fd);
 
@@ -314,8 +316,8 @@ void pixel (int x, int y, unsigned colidx)
        unsigned xormode;
        union multiptr loc;

-       if ((x < 0) || ((__u32)x >= var.xres_virtual) ||
-           (y < 0) || ((__u32)y >= var.yres_virtual))
+       if ((x < 0) || ((uint32_t)x >= var.xres_virtual) ||
+           (y < 0) || ((uint32_t)y >= var.yres_virtual))
                return;
 
        xormode = colidx & XORMODE;
@@ -386,10 +388,10 @@ void fillrect (int x1, int y1, int x2, int y2, unsigned colidx)
        /* Clipping and sanity checking */
        if (x1 > x2) { tmp = x1; x1 = x2; x2 = tmp; }
        if (y1 > y2) { tmp = y1; y1 = y2; y2 = tmp; }
-       if (x1 < 0) x1 = 0; if ((__u32)x1 >= xres) x1 = xres - 1;
-       if (x2 < 0) x2 = 0; if ((__u32)x2 >= xres) x2 = xres - 1;
-       if (y1 < 0) y1 = 0; if ((__u32)y1 >= yres) y1 = yres - 1;
-       if (y2 < 0) y2 = 0; if ((__u32)y2 >= yres) y2 = yres - 1;
+       if (x1 < 0) x1 = 0; if ((uint32_t)x1 >= xres) x1 = xres - 1;
+       if (x2 < 0) x2 = 0; if ((uint32_t)x2 >= xres) x2 = xres - 1;
+       if (y1 < 0) y1 = 0; if ((uint32_t)y1 >= yres) y1 = yres - 1;
+       if (y2 < 0) y2 = 0; if ((uint32_t)y2 >= yres) y2 = yres - 1;
 
        if ((x1 > x2) || (y1 > y2))
                return;

merge, 2017-03-20:
That's great. I only waited until this not using specific types causes trouble :). Definitely go for your second approach, this is Linux specific after all! This fixes the problem for you? I wonder why I can't reproduce it on amd64...

Anyways, here's what we're gonna do: Since this affects all of Linux, I won't put it in 1.8. I want to release that today. I'll do a 1.9-rc1 very shortly thereafter, when this is ready (and maybe I do some additions to the input raw module).

Minor notes:

I think you can include <stdint.h> in fbutls.h (instead of <sys/stdin.h> that is included there now), and leave it out on the c file.

Would you mind making a real pull request of your working state? If you do, please make seperate commits for formatting cleanup. Try to do a minimal necessary commit that fixes your problem, and then whatever bonus you want to have in seperate commits.

neilwhelchel, 2017-03-21:
Yes, either patch above fixes the framebuffer issue on both the x86_64 and aarch64.
The same issue is present on x86_64, but only with a 32 BPP framebuffer. It is not present with a 24 BPP framebuffer (the most common) because the original code uses char (3 times) to write out 24 bits. Where the code works as expected with 32 bit framebuffers is on 32 bit machines (where long is 32 bits). On most 64 bit machines long is 64 bits, and if the framebuffer is 32 bits, the code fails because it casts the pointer to the framebuffer to a long when it detects that the framebuffer is 32 bits.

Sorry about the messy approach I took to handle this, the original plan was to point you in the right direction and rely on your judgement as to the best approach to fix this issue given your familiarity and experience with the existing code. Next thing I knew I was submitting patches instead of pull requests. I will correct this issue.

merge, 2017-03-21:
As much as I want to discuss this publically, if you don't want to use github so much, I also accept git send-email patches at martink@posteo.de . There shouldn't be a barrier for you. (We don't have a mailing list. I'm currently asking around where I could have a list hosted).

However. I'm looking forward to reviewing and testing your patches on my devices!

thanks a lot!

neilwhelchel, 2017-03-21:
That is not what I meant. I meant that I am happy to use GitHub, and I should have started out doing so. When I started, I was thinking that my involvement was to be trivial, but now, I am far more involved than I anticipated. That is a good thing. ;)
Also, I have a server that I would offer for the mailing list.

merge, 2017-03-21:
The more involvement the better. I'd happily not be the lead developer alone here :)

Thanks for the offer of a server! I'll wait a little, but may get back to you via email regarding this.

merge, 2017-03-25:
We already got a mailing list. Thanks again for your offer though. Are you still working on a fix here? If not, that's fine too. Just let me know. Thanks

merge, 2017-03-28:
I guess I'll go ahead and apply your changes roughly myself, tomorrow or thursday. If you're working on a pull request, just let me know and I'll wait :)

commit 73773ac9 (2017-04-04): fbutils-linux: fix framebuffer errors on 64 bit systems

... by using defined standard datatypes instead of short, int and long.
Also, __u32 is replaced by uint32_t.

Fixes #85
closed 2017-04-04 by merge