libts/tslib #88: Error on cross compiler

Moonbeem, 2017-03-22:
I'm trying to cross compile tslib on Ubuntu16.04_i386 with arm-linux-gnueabihf compiler.
I launch config/compile with:

export CC=/usr/local/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-gcc
export CXX=/usr/local/arm-linux-gnueabihf/bin/arm-linux-gnueabihf-g++
./autogen.sh
./configure --host=arm-linux-gnueabihf ac_cv_func_malloc_0_nonnull=yes --cache-file=arm-linux.cache --prefix /usr/local/tslib 
make

Error：
libtool: link:  /....................................................../
../src/.lib/libts.so:  undefined reference to 'dlopen'
../src/.lib/libts.so:  undefined reference to 'dlclose'
../src/.lib/libts.so:  undefined reference to 'dlsym'
collect2: error : ld returned 1 exit status

Thanks.

merge, 2017-03-22:
has building tslib building this way worked before? Do you have libdl.so.* installed? Do you successfully cross compile other programs? We (and buildroot) don't have problems cross compiling...

merge, 2017-04-10:
I assume the issue with compiling wasn't on our side or is fixed for you. I clean up and close here, but please feel free to re-open if you can't build tslib. thanks!

closed 2017-04-10 by merge

zhenguoli, 2024-07-23:
When you built with configure script, it has `-fvisibility` set when building, so the symbols are local which cannot be linked. With current tslib version, cmake build works well.