libts/tslib #89: properly draw our logo

merge, 2017-03-23:
our logo is basically a draft outline :) If you have the digital drawing skills to create a beautiful proper SVG file following the idea of the current one, _please_ let us know!

rrcha, 2017-11-10:
Not the best at vector graphics in the world, but I had a go :)   Let me know if you would like the SVG.  Saw the request on the pmOS forums.

[image]

merge, 2017-11-10:
That's awesome! Thanks for doing this. Could I ask for a 2nd version with basicaly 2 changes?

* lines of hand and letters the same strength
* background raster removed

and do _you_ like it? Any idea for a change in general?

cool

craftyguy, 2017-11-10:
I'm not the (magnificent!) submitter of this, but I quite like the logo!

merge, 2017-11-10:
And yes, if I were to replace the logo, I'd only want to do it with an SVG image.

rrcha, 2017-11-10:
I've made a few changes per request as attached:

[image]

As per the design, yes, I like it!  I can't think of any changes I'd make.

As for the favicon, maybe something like this?

[image]

merge, 2017-11-10:
I like it :) If you'll provide the svg files and permissions to resize and use them as fit, I guess I'll take them.

What do others think?

rrcha, 2017-11-10:
Sure. I've [uploaded the vector (and raster) versions of the file here](https://github.com/rrcha/TSLIB-logo) and waived all rights to the file.

It would be nice to get a mention in your contributors list though!

merge, 2017-11-10:
Oh sure. Please give me your name and email address in order to add you to the THANKS file.

rrcha, 2017-11-10:
Ravi Chandarana (ravichandarana@rocketmail.com)

Many thanks!

merge, 2017-11-10:
One more thing: Would you resize the logo to be smaller for me? About 2/3 of the current size maybe. I just saw that it's not as easy as I thought -.-

You can push a commit to your repository, overwriting the svg file of course.

thanks!

rrcha, 2017-11-10:
Ok, done!

merge, 2017-11-10:
Wow, I really think this is totally awesome! If you ask me this improves the project quite a bit. This is a pretty high quality product and now we have a logo that can represent it properly.

thanks!

closed 2017-11-10 by merge

rrcha, 2017-11-10:
My pleasure.  Best wishes for the project!

commit 814aef9b (2017-11-10): Replace the logo with a shiny new svg logo file

See https://github.com/rrcha/TSLIB-logo to the sources and thanks a lot
to Ravi for doing this work. We create a professional quality product and
now have a logo that can way better represent our project.

This is an awesome improvement for our project!

Fixes #89