libts/tslib #90: Support Sitronix ST1232

avazquezdev, 2017-03-29:
Hi. I have a touchscreen (Sitronix ST1232 ) that don't support BTN_TOUCH and BTN_LEFT

This touchscreen only support next event:
/dev/input/event0:      st1232-touchscreen
Select the device event number [0-0]: 0
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "st1232-touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
  Event type 3 (EV_ABS)
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max      255
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      799
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max      479

Would it be possible to support this touch panel?

Thanks!
Regards!

merge, 2017-03-29:
Please paste a complete output of evtest in here, doing a short one-finger swipe. We'll see clearer then.

avazquezdev, 2017-03-30:
Done. Thanks for the help!

evtest
...
Properties:                                                                     
Testing ... (interrupt to exit)                                                 
Event: time 16.690766, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.690766, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 113  
Event: time 16.690766, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 395  
Event: time 16.690766, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.690766, -------------- SYN_REPORT ------------                   
Event: time 16.699848, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.699848, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 113  
Event: time 16.699848, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 395  
Event: time 16.699848, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.699848, -------------- SYN_REPORT ------------                   
Event: time 16.708780, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.708780, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 113  
Event: time 16.708780, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 395  
Event: time 16.708780, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.708780, -------------- SYN_REPORT ------------                   
Event: time 16.717821, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.717821, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 113  
Event: time 16.717821, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 395  
Event: time 16.717821, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.717821, -------------- SYN_REPORT ------------                   
Event: time 16.726925, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.726925, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 113  
Event: time 16.726925, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 394  
Event: time 16.726925, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.726925, -------------- SYN_REPORT ------------                   
Event: time 16.736032, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.736032, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 113  
Event: time 16.736032, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 393  
Event: time 16.736032, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.736032, -------------- SYN_REPORT ------------                   
Event: time 16.747655, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.747655, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 114  
Event: time 16.747655, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 391  
Event: time 16.747655, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.747655, -------------- SYN_REPORT ------------                   
Event: time 16.759301, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.759301, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 114  
Event: time 16.759301, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 391  
Event: time 16.759301, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.759301, -------------- SYN_REPORT ------------                   
Event: time 16.770941, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.770941, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 114  
Event: time 16.770941, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 391  
Event: time 16.770941, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.770941, -------------- SYN_REPORT ------------                   
Event: time 16.782534, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.782534, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 114  
Event: time 16.782534, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 389  
Event: time 16.782534, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.782534, -------------- SYN_REPORT ------------                   
Event: time 16.794137, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.794137, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 114  
Event: time 16.794137, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 389  
Event: time 16.794137, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.794137, -------------- SYN_REPORT ------------                   
Event: time 16.805799, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.805799, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 114  
Event: time 16.805799, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 389  
Event: time 16.805799, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 16.805799, -------------- SYN_REPORT ------------                   
Event: time 16.817359, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 38  
Event: time 16.817359, type 3 (EV_ABS[   18.269146] random: nonblocking pool isd
), code 53 (ABS_MT_POSITION_X), value 114                                       
Event: time 16.817359, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 389  
[... 84 lines cut ...]
Event: time 17.032397, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 383  
[... cut ...]
Event: time 17.077134, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 42  
Event: time 17.077134, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 114  
Event: time 17.077134, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 383  
Event: time 17.077134, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 17.077134, -------------- SYN_REPORT ------------                   
Event: time 17.084241, ++++++++++++++ SYN_MT_REPORT ++++++++++++                
Event: time 17.084241, -------------- SYN_REPORT ------------

merge, 2017-03-30:
thanks a lot! Yes, we have to fix supporting it in tslib.

I added an improvement to the master branch now, which would be awesome if you could pull and test. How do you test? Do you run ts_calibrate to see if you can use it? (which is totally fine, I'm just curious).

Furthermore, if you still have problems using tslib after that, please also paste an `evtest` output of a short **two**-finger swipe.

again, thanks for taking the time to report and help.

merge, 2017-03-30:
I had forgotten to handle ts_read() too. as of now you have a chance that ts_calibrate will work. sorry for writing too early before.

avazquezdev, 2017-03-30:
I had to comment the part of the code where it detects if the events BTN_TOUCH or BTN_LEFT are disponible because as the controller doesn't have these events it doesn't detect the touch panel. Message Error:
> Selected device is not a touchscreen (must support BTN_TOUCH or BTN_LEFT events)

Then I calibrated and executed the test and it works correctly.

It seems that multitactile events don't work well but this is driver problem. I will continue investigating.

Thanks!

merge, 2017-03-30:
sure. I forgot to adapt the device check. Thanks a lot. I'll correct that soon!

Multitouch, if not working correctly in ts_test_mt, should also be fixed in tslib. It's a little hard for me, as I don't have the device myself and can't test.

merge, 2017-03-30:
device checking is fixed in input-raw.

Just describe, how multitouch feels wrong, using ts_test_mt. I can imagine that we don't support your device correctly.

merge, 2017-03-30:
If you want to give me a name and an email address, I'll add you to the THANKS file

avazquezdev, 2017-03-31:
> Just describe, how multitouch feels wrong, using ts_test_mt. I can imagine that we don't support your device correctly.
Yes. ts_test_mt/ts_print_mt only detect 1 slot. I paste an evtest output of a short two-finger swipe

Testing ... (interrupt to exit)
Event: time 1167746354.082448, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.082448, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 122
Event: time 1167746354.082448, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 104
Event: time 1167746354.082448, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.082448, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.082448, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 349
Event: time 1167746354.082448, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 203
Event: time 1167746354.082448, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.082448, -------------- SYN_REPORT ------------
Event: time 1167746354.083769, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.083769, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 122
Event: time 1167746354.083769, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 104
Event: time 1167746354.083769, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.083769, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.083769, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 349
Event: time 1167746354.083769, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 203
Event: time 1167746354.083769, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.083769, -------------- SYN_REPORT ------------
Event: time 1167746354.093566, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.093566, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 122
Event: time 1167746354.093566, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 104
Event: time 1167746354.093566, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.093566, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.093566, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 349
Event: time 1167746354.093566, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 203
Event: time 1167746354.093566, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.093566, -------------- SYN_REPORT ------------
Event: time 1167746354.094882, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.094882, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 122
Event: time 1167746354.094882, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 104
Event: time 1167746354.094882, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.094882, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.094882, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 349
Event: time 1167746354.094882, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 203
Event: time 1167746354.094882, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.094882, -------------- SYN_REPORT ------------
Event: time 1167746354.104329, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.104329, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 122
Event: time 1167746354.104329, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 105
Event: time 1167746354.104329, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.104329, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.104329, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 349
Event: time 1167746354.104329, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 203
Event: time 1167746354.104329, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.104329, -------------- SYN_REPORT ------------
Event: time 1167746354.105640, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.105640, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 122
Event: time 1167746354.105640, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 105
Event: time 1167746354.105640, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.105640, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.105640, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 349
Event: time 1167746354.105640, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 203
Event: time 1167746354.105640, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.105640, -------------- SYN_REPORT ------------
Event: time 1167746354.115477, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.115477, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 121
Event: time 1167746354.115477, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 106
Event: time 1167746354.115477, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.115477, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 11
Event: time 1167746354.115477, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 349
Event: time 1167746354.115477, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 203
Event: time 1167746354.115477, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.115477, -------------- SYN_REPORT ------------
[... cut ...]
Event: time 1167746354.149643, -------------- SYN_REPORT ------------
Event: time 1167746354.170508, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.170508, -------------- SYN_REPORT ------------
Event: time 1167746354.171816, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1167746354.171816, -------------- SYN_REPORT ------------

> If you want to give me a name and an email address, I'll add you to the THANKS file
Ok. Alex Vazquez --> avazquez.dev@gmail.com

Thanks!!
Regards!!

merge, 2017-03-31:
could you pull and build again and post the output of ts_print_mt if it doesn't work?

avazquezdev, 2017-04-01:
> could you pull and build again and post the output of ts_print_mt touching with 2 fingers, if it doesn't work?

No problem, I test this the monday.

avazquezdev, 2017-04-03:
> could you pull and build again and post the output of ts_print_mt touching with 2 fingers, if it doesn't work?
Ok. it doesn't work.

ts_print_mt

sample 0 - 1491220233.615828 - (slot 0)    732     68    255
sample 0 - 1491220233.623629 - (slot 0)    732     68    255
sample 0 - 1491220233.638221 - (slot 0)    732     68    255
sample 0 - 1491220233.652952 - (slot 0)     39    357    255
sample 0 - 1491220233.660229 - (slot 0)     38    356    255
sample 0 - 1491220233.660229 - (slot 0)     38    354      0

merge, 2017-04-03:
ah. we needed an override for slots (concurrent touch contacts). The good thing is, we needed that anyways, not "only" because of devices like yours: if you pull and build again, you can test `ts_print_mt -j 2` or `ts_test_mt -j 2` to have 2 possible touch contacts. But I'd test `ts_test_mt -j 10` (or more if you're creative :) ), to see how many concurrent touch contacts your device driver can handle.

... if I didn't make a mistake implementing this all blindly...

merge, 2017-04-03:
don't forget, *not* to use the variance or skip filter module.

merge, 2017-04-04:
update: your driver (device) can handle only 2 fingers maximum.

merge, 2017-04-10:
I assume this is fixed for you. If not, please feel free to re-open this or open a new issue. In order to clean up I go ahead and close this.

That said, I'd love to get more feedback from this or similar devices.

closed 2017-04-10 by merge