libts/tslib #91: "undefined reference to `ts_open'"

ixuhan, 2017-04-07:
Hello,The tslib give me a great help.
The command "ts_calibrate","ts_test" and "ts_print" can normal work.
Then i write a demo,but it don’t work for me .
it shows like this,

- /tmp/ccMqz0EN.o: In function `getTsXY':
- test.c:(.text+0x14): undefined reference to `ts_open'
- test.c:(.text+0x3c): undefined reference to `ts_config'
- test.c:(.text+0x6c): undefined reference to `ts_read'
- collect2: error: ld returned 1 exit status

My code :https://github.com/ixuhan/hello-world/blob/master/tslibTest.c

Thanks.

merge, 2017-04-08:
what version of tslib do you use?

merge, 2017-04-08:
do you tell gcc to link against libts?

closed 2017-04-10 by ixuhan

zhenguoli, 2024-07-23:
When you built with configure script, it has `-fvisibility=hidden` set when building, so the symbols are local which cannot be linked. With current tslib version, cmake build works well. Refer https://stackoverflow.com/questions/37531846/nm-symbol-output-t-vs-t-in-a-shared-so-library