libts/tslib #92: ts_uinput starts only from prompt properly

vitalogy, 2017-04-25:
Servus,

ts_uinput starts not properly from a udev rule-file or from systemd service-file.

**/etc/udev/rules.d/98-touchscreen.rules**
```
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="gsl3675", SYMLINK+="input/ts_raw"
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="ts_uinput", SYMLINK+="input/ts_uinput"
```
Udev will link the ```/dev/input/ts_raw```to the touchscreen event ```/dev/input/eventX``` and will create ```/dev/input/ts_uinput``` when a event-file exist with the name ```ts_uinput``` created by the executable ts_uinput.

Running the following script directly from a prompt works, but not when invoked from the udev rule by RUN+="/usr/local/bin/ts_uinput.sh" or started from a systemd service-file.
ts_uinput gives some feedback in journal but does not create the eventX under /dev/input. 

**/usr/local/bin/ts_uinput.sh**
```
#!/bin/sh

export TSLIB_TSEVENTTYPE=INPUT
export TSLIB_TSDEVICE=/dev/input/ts_raw
export TSLIB_CALIBFILE=/etc/pointercal
export TSLIB_CONFFILE=/etc/ts.conf
export TSLIB_PLUGINDIR=/usr/lib/ts
export TSLIB_FBDEVICE=/dev/fb0
export TSLIB_CONSOLEDEVICE=none

/usr/bin/ts_uinput -d

echo 8 > /sys/class/backlight/backlight/brightness
echo 0 > /sys/class/backlight/backlight/bl_power
```

**/usr/lib/systemd/system/ts_uinput.service**
```
[Unit]
Description=touchscreen input

[Service]
Type=simple
ExecStart=/usr/local/bin/ts_uinput.sh

[Install]
WantedBy=multi-user.target
```

running the script with systemd: **systemctl start ts_uinput**
```
# systemctl start ts_uinput

# journalctl -r
<snip>
Apr 25 23:45:53 openHAB ts_uinput.sh[724]: input3
Apr 25 23:45:53 openHAB kernel: input: ts_uinput as /devices/virtual/input/input3
</snip>

# ll /dev/input/
total 0
drwxr-xr-x  3 root root     180 25. Apr 23:45 .
drwxr-xr-x 13 root root    3,0K 25. Apr 23:19 ..
drwxr-xr-x  2 root root     100 25. Apr 23:19 by-path
crw-rw----  1 root input 13, 64 25. Apr 23:19 event0
crw-rw----  1 root input 13, 65 25. Apr 23:19 event1
crw-rw----  1 root input 13, 66 25. Apr 23:19 event2
crw-rw----  1 root input 13, 63 25. Apr 23:19 mice
crw-rw----  1 root input 13, 32 25. Apr 23:19 mouse0
lrwxrwxrwx  1 root root       6 25. Apr 23:19 ts_raw -> event2
```

running the script from prompt works:
```
# /usr/local/bin/ts_uinput.sh 
input4

# journalctl -r
<snip>
Apr 25 23:49:21 openHAB kernel: input: ts_uinput as /devices/virtual/input/input4
</snip>

# ll /dev/input/
total 0
drwxr-xr-x  3 root root     240 25. Apr 23:49 .
drwxr-xr-x 13 root root    3,0K 25. Apr 23:19 ..
drwxr-xr-x  2 root root     100 25. Apr 23:19 by-path
crw-rw----  1 root input 13, 64 25. Apr 23:19 event0
crw-rw----  1 root input 13, 65 25. Apr 23:19 event1
crw-rw----  1 root input 13, 66 25. Apr 23:19 event2
crw-rw----  1 root input 13, 67 25. Apr 23:49 event3
crw-rw----  1 root input 13, 63 25. Apr 23:19 mice
crw-rw----  1 root input 13, 32 25. Apr 23:19 mouse0
crw-rw----  1 root input 13, 33 25. Apr 23:49 mouse1
lrwxrwxrwx  1 root root       6 25. Apr 23:19 ts_raw -> event2
lrwxrwxrwx  1 root root       6 25. Apr 23:49 ts_uinput -> event3
```

/dev/input/event3 was created and udev could create a symlink to /dev/input/ts_uinput

Is there a chance to get this automatically running at system startup?

merge, 2017-04-25:
I'm automatically starting ts_uinput on our devices at startup, but only using rc.d scripts. So I hope we can resolve this for you using udev too.

I'm really not sure now. maybe some return value issue with the bash script? I don't really know udev well, but maybe you can compare your script to others that work with udev.

Hm. You could temporarily try `ts_uinput -v &` in your script, just to get some tslib output in your logs.

I could try this on my laptop, but I can't promise when I'll get to playing with it.

merge, 2017-04-26:
Maybe you have to set the environment variables for udev differently? There is a key like `ENV{some_var}="value"` for udev rules.

vitalogy, 2017-04-27:
okay, got it:

**udev rule**
/etc/udev/rules.d/98-touchscreen.rules
```
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{phys}=="input/ts", SYMLINK+="input/ts_raw", TAG+="systemd"
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="ts_uinput", SYMLINK+="input/ts_uinput"
```
Udev will create a symlink ```/dev/input/ts_raw```to the device tagged with ```ATTRS{phys}=="input/ts"```. With ```TAG+="systemd"``` the input-device units (eventX and ts_raw) will show up in systemd whenever the device is available. But we will only use the ```dev-input-ts_raw.device```.

**systemd service**
Create a file with the environment variables for ts_uinput.
/etc/ts.env
```
TSLIB_TSEVENTTYPE=INPUT
TSLIB_TSDEVICE=/dev/input/ts_raw
TSLIB_CALIBFILE=/etc/pointercal
TSLIB_CONFFILE=/etc/ts.conf
TSLIB_PLUGINDIR=/usr/lib/ts
TSLIB_FBDEVICE=/dev/fb0
TSLIB_CONSOLEDEVICE=none
```

Create the systemd service file for ts_uinput.
/usr/lib/systemd/system/ts_uinput.service
```
[Unit]
Description=touchscreen input
Wants=dev-input-ts_raw.device
After=dev-input-ts_raw.device

[Service]
Type=oneshot
EnvironmentFile=/etc/ts.env
ExecStart=/bin/sh -c 'exec /usr/bin/ts_uinput -v &> /var/log/ts_uinput.log'

[Install]
WantedBy=multi-user.target
```
Enable the service with ```#systemctl enable ts_uinput```  and reboot.
The service ```ts_uinput.service``` will start when the device ```dev-input-ts_raw.device``` will be show up in systemd as a input-device unit. Then udev will create a symlink ```/dev/input/ts_uinput``` to the device tagged with ```ATTRS{name}=="ts_uinput"```, created by ts_uinput. Thats it.

But it surprise me, that ts_uinput only starts when it is redirect to something, here a log file in ```/var/log``` ... but the log file is empty, although ts_uinput was started with ```-v``` .

merge, 2017-04-27:
ok. Starting `ts_uinput -v &` was just a debugging suggestion. Again I'm not sure why redirection doesn't work in that case :) But does your solution still work, if you use `ts_uinput -d` again? I guess you can directly say `ExecStart=/usr/bin/ts_uinput -d`

vitalogy, 2017-04-27:
not really, probably i am doing something wrong
```
# systemctl status ts_uinput
● ts_uinput.service - touchscreen input
   Loaded: loaded (/usr/lib/systemd/system/ts_uinput.service; enabled; vendor preset: disabled)
   Active: inactive (dead) since Thu 2017-04-27 18:38:49 CEST; 2s ago
  Process: 695 ExecStart=/usr/bin/ts_uinput -d (code=exited, status=0/SUCCESS)
 Main PID: 695 (code=exited, status=0/SUCCESS)

Apr 27 18:38:49 openHAB systemd[1]: Starting touchscreen input...
Apr 27 18:38:49 openHAB ts_uinput[695]: input5
Apr 27 18:38:49 openHAB systemd[1]: Started touchscreen input.
root@openHAB /home/michael # ll /dev/input/
total 0
drwxr-xr-x  3 root root     180 27. Apr 18:38 .
drwxr-xr-x 13 root root    3,0K 27. Apr 18:36 ..
drwxr-xr-x  2 root root     100 27. Apr 18:36 by-path
crw-rw----  1 root input 13, 64 27. Apr 18:36 event0
crw-rw----  1 root input 13, 65 27. Apr 18:36 event1
crw-rw----  1 root input 13, 66 27. Apr 18:36 event2
crw-rw----  1 root input 13, 63 27. Apr 18:36 mice
crw-rw----  1 root input 13, 32 27. Apr 18:36 mouse0
lrwxrwxrwx  1 root root       6 27. Apr 18:36 ts_raw -> event2
```

merge, 2017-04-27:
btw. `-v` prints touch points - way too much for productive use. If you have to work around it somehow, be sure to *not* use `-v`

merge, 2017-04-28:
does `ExecStart=/usr/bin/ts_uinput -d &> /var/log/ts_uinput.log` work? I wonder if ts_uinput could be changed to support your usage better...

vitalogy, 2017-04-28:
The one that is working for me: 
```ExecStart=/bin/sh -c 'exec /usr/bin/ts_uinput &> /var/log/ts_uinput.log'```

These are failing (ts_uinput starts, but do not run):
```ExecStart=/usr/bin/ts_uinput -d```
```ExecStart=/usr/bin/ts_uinput -d &> /var/log/ts_uinput.log```
```ExecStart=/bin/sh -c 'exec /usr/bin/ts_uinput -d'```
```ExecStart=/bin/sh -c 'exec /usr/bin/ts_uinput -d &> /var/log/ts_uinput.log'```

merge, 2017-04-28:
It's perfectly fine how you use it.

Unfortunately I don't know udev well enough, but it might expect the program not to return. Isn't udev itself a daemon? So, while the full explanation is missing here, ts_uinput is totally correctly used.

What's more interesting to me is, why the output redirection is needed. If you're happy though, feel free to close this issue. I'll think about adding something of it to the README.

And if you want to, give me your name and email address and I'll add you to the THANKS file.

so, thanks.

merge, 2017-05-04:
hm. So while I feel one could understand things better with your setup than I do, I assume you're fine with your solution. Again, there's also nothing wrong with it, and also because I added your setup to the README, I'll close this issue in order to clean up.

You know, in case you're still working on it, or have problems, really just reopen or write another one. thanks!

closed 2017-05-04 by merge

nap0nap, 2019-01-29:
After few try, this one is working for me :

cat /etc/udev/rules.d/98-touchscreen.rules 
```
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="eGalax Inc. USB TouchController Pen", SYMLINK+="input/touchscreen", TAG+="systemd"
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="ts_uinput", SYMLINK+="input/touchscreen_ts_uinput"
```

```

[Unit]
Description=touchscreen input
Wants=dev-input-touchscreen.device  #HERE it is dev-input-DEVICENAME_IN_UDEV_RULES.device
After=dev-input-touchscreen.device  #HERE it is dev-input-DEVICENAME_IN_UDEV_RULES.device

[Service]
Type=forking	# IF THE PROCESS CHANGE IS PID INSTEAD OF 'simple', ELSE SYSTEMD KILL IT
EnvironmentFile=/etc/ts.env
ExecStart=/bin/sh -c 'exec env LD_LIBRARY_PATH=/usr/local/lib /usr/local/bin/ts_uinput -d'   #env AFTER exec (because I don't use default path)
[Install]
WantedBy=multi-user.target
```

By the way, thanks for your works !

nap0nap, 2019-01-29:
And the best way to minimize forking is :

cat /etc/udev/rules.d/98-touchscreen.rules
```
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="eGalax Inc. USB TouchController Pen", SYMLINK+="input/touchscreen", TAG+="systemd"
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="ts_uinput", SYMLINK+="input/touchscreen_ts_uinput"
```

cat /etc/ts.env 
```
LD_LIBRARY_PATH=/usr/local/lib
TSLIB_PLUGINDIR=/usr/local/lib/ts
TSLIB_CONFFILE=/usr/local/etc/ts.conf
TSLIB_CALIBFILE=/etc/pointercal
TSLIB_TSEVENTTYPE=INPUT
TSLIB_TSDEVICE=/dev/input/touchscreen
TSLIB_CALIBFILE=/etc/pointercal
TSLIB_FBDEVICE=/dev/fb0
TSLIB_CONSOLEDEVICE=none
```

cat /usr/lib/systemd/system/ts_uinput.service
```
  [Unit]
  Description=touchscreen input
  Wants=dev-input-touchscreen.device
  After=dev-input-touchscreen.device

  [Service]
  Type=forking
  EnvironmentFile=/etc/ts.env
  ExecStart=/usr/local/bin/ts_uinput -d
  [Install]
  WantedBy=multi-user.target
```

merge, 2019-01-29:
thanks for your solution! do you think we should improve our docs?

nap0nap, 2019-01-29:
_Yes, it can be a good things.
Somethings like (with ts_raw as name of source device link) :_

if you're using systemd, create the following udev rule, for example /etc/udev/rules.d/98-touchscreen.rules:

```
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="NAME_OF_THE_TOUCH_CONTROLLER", SYMLINK+="input/ts_raw", TAG+="systemd"
SUBSYSTEM=="input", KERNEL=="event[0-9]*", ATTRS{name}=="ts_uinput", SYMLINK+="input/ts_uinput"
```

NAME_OF_USB_TOUCH_CONTROLLER is returned with :

` cat /proc/bus/input/devices | grep Name`

running as systemd service (optional)

in case you have to use non-default paths, create a file containing the environment for tslib, like /etc/ts.env
```
LD_LIBRARY_PATH=/usr/local/lib
TSLIB_PLUGINDIR=/usr/local/lib/ts
TSLIB_CONFFILE=/usr/local/etc/ts.conf
TSLIB_CALIBFILE=/etc/pointercal
TSLIB_TSEVENTTYPE=INPUT   #I don't know if it's needed
TSLIB_TSDEVICE=/dev/input/ts_raw  #I don't know if it's needed
TSLIB_FBDEVICE=/dev/fb0       #I don't know if it's needed
TSLIB_CONSOLEDEVICE=none  #I don't know if it's needed
```

and create a systemd service file, like /usr/lib/systemd/system/ts_uinput.service

```
  [Unit]
  Description=touchscreen input
  Wants=dev-input-ts_raw.device
  After=dev-input-ts_raw.device

  [Service]
  Type=forking
  EnvironmentFile=/etc/ts.env
  ExecStart=/usr/local/bin/ts_uinput -d
  [Install]
  WantedBy=multi-user.target
```

_2 things are important in in the systemd service file :_ 
 _- Property "Wants" and "After" depends of the name of the SYMLINK in udev rule._
 _- Type=forking because the PID of the process change._

_I let you translate that at best, is you have any question, I'm here._