libts/tslib #95: Extraneous leading sample inserted by median filter

ClayMontgomery, 2017-05-03:
I'm seeing an issue with the median filter. It usually inserts a leading sample with uninitialized x and y values at the start of each set of samples from each touch gesture. It does this most of the time, but not always. The pressure value is always 0 and the x and y values are garbage, but always the same pair of values. I understand x and y should not be valid when the pressure is 0, but with other filters, the 0 pressure sample only occurs once at the end of a sequence of samples. With the median filter, the 0 pressure sample occurs at the start and again at the end of each gesture sequence. I am calling ts_read_mt() in non-blocking mode and always requesting 10 samples. I usually get between 1 to 5 samples per call.  I am using tslib 1.9 and the complete list of filters I am using are:

module_raw input
module pthres pmin=1
module median depth=6
module linear

Regards, Clay

merge, 2017-05-04:
You're right. Actually it should be depth/2 of those zero-samples. In your case actually 5/2 (median only works for odd numbers depths), which becomes 2. So two, not only one. Is that correct?

Thanks for reporting this! It can't indeed be the intended behaviour. I'll have a look as time permits, but I'm confident that this will be fixed for the 1.10 release.

merge, 2017-05-04:
Please re-open if you can test and still have problems with it. And if you want to, give me your email address to put in the THANKS file. so, thanks.

closed 2017-05-04 by merge

merge, 2017-05-04:
just fyi, there's already the 1.10-rc1 tarball release in case you want to test that fix early.

commit 5974266e (2017-05-04): median: read_mt: mark all-zero samples invalid

the median filter "waits" depth/2 samples until it has enough data to do
it's math. The first depth/2 samples after pen-down aren't valid. We set
them to zero, but actually served them to the user as valid samples.

This fixes this behaviour for ts_read_mt() only.

Fixes #95