libts/tslib #96: master: error - ts_read_mt

hawkeyexp, 2017-05-14:
I have done a compile from branch master and get the following error with ts_uinput:
(lib paths are correct cause i moved/linked libraries)

sudo TSLIB_FBDEVICE=/dev/fb0 TSLIB_TSDEVICE=/dev/input/event3 TSLIB_CALIBFILE=/etc/pointercal TSLIB_CONFFILE=/etc/ts.conf TSLIB_PLUGINDIR=/usr/lib/arm-linux-gnueabihf/ts ts_uinput -v -d
ts_uinput: symbol lookup error: ts_uinput: undefined symbol: ts_read_mt

ts_calbrate is working fine.

I'm trying to get eGalax display working on rasbian with kodi standalone but no luck.

Also tried release 1.10 but on this build ts_calibrate wont give me any result for pointercal (result always 0).

merge, 2017-05-14:
what's in your ts.conf?
does ts_test work? ts_test_mt?

hawkeyexp, 2017-05-14:
ts_test and ts_calibration are working perfect - ts_test_mt not checked cause of heavy frustration :-)

merge, 2017-05-14:
again, your ts.conf and your ./configure parameters would be interesting

merge, 2017-05-14:
and please try leaving all variables out when running ts_uinput. What does "ts_uinput -v" say?

merge, 2017-05-14:
TSLIB_TSDEVICE should be the only variable you need.

And it seems to me you have problems with your installation. Try to uninstall or delete all of libts and install it cleanly.

closed 2017-05-15 by hawkeyexp