libts/tslib #97: Synaptics rmi4 (sensor00fn11) doesn't support BTN_TOUCH or BTN_LEFT

M1cha, 2017-05-14:
- error message:
```
tslib: Selected device is not a touchscreen (missing BTN_TOUCH or BTN_LEFT)
```
- platform: apq8064 with a 3.4.0 kernel
- touchscreen driver: https://github.com/LineageOS/android_kernel_xiaomi_aries/tree/cm-11.0/drivers/input/rmi4
- ts.conf:
```
module_raw input
module linear
```
- evtest:
```
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "sensor00fn11"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
  Event type 3 (EV_ABS)
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max       15
    Event code 49 (ABS_MT_TOUCH_MINOR)
      Value      0
      Min        0
      Max       15
    Event code 52 (ABS_MT_ORIENTATION)
      Value      0
      Min        0
      Max        1
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max     1100
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     1900
    Event code 55 (ABS_MT_TOOL_TYPE)
      Value      0
      Min        0
      Max        0
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
    Event code 58 (ABS_MT_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
Testing ... (interrupt to exit)
Event: time 1314.720628, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 72
Event: time 1314.720628, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 84
Event: time 1314.720628, type 3 (EV_ABS), code 49 (ABS_MT_TOUCH_MINOR), value 4
Event: time 1314.720628, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 889
Event: time 1314.720628, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1163
Event: time 1314.720628, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1314.720659, -------------- SYN_REPORT ------------
Event: time 1314.790270, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1314.790270, -------------- SYN_REPORT ------------
Event: time 1315.190263, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 73
Event: time 1315.190263, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 83
Event: time 1315.190294, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 3
Event: time 1315.190294, type 3 (EV_ABS), code 49 (ABS_MT_TOUCH_MINOR), value 3
Event: time 1315.190294, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 332
Event: time 1315.190294, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 1077
Event: time 1315.190294, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1315.190294, -------------- SYN_REPORT ------------
Event: time 1315.250139, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 1315.250139, -------------- SYN_REPORT ------------
```

merge, 2017-05-14:
what version of tslib are you using? Thanks!

M1cha, 2017-05-14:
@merge thx for the fast replay :) I'm using the latest master at 5d2fbf9

merge, 2017-05-14:
While this is fine now, remember, master is not meant to be stable. only releases are, at least supposed to be.

At first sight this is something we can and should fix, but I'll be able to look at it in detail sometime during the upcoming week.

merge, 2017-05-14:
if you can, try deleting this check for BTN_TOUCH in input-raw.c's check_fd(). you'll find it by looking for the printf printing your error.

M1cha, 2017-05-14:
that seems to work. Is there a way I can check for bad side-effects or miscalculations?

merge, 2017-05-14:
I think it's ok and this will be the fix here :) I'll have to think about it just a little more, run it on a few devices and so on. But it'll be fixed for 1.11 and you can send a pull request if you have verified it wotks for you, if you want to.

Thanks for reporting

closed 2017-05-15 by merge

merge, 2017-05-15:
If you want thank, give me your name and email address and I can add you to the THANKS file.

M1cha, 2017-05-17:
@merge did you delete my comment or did I forget to hit the send button?

"Michael Zimmermann <sigmaepsilon92@gmail.com>"

merge, 2017-05-17:
didn'nt knowingly delete it. got it now ;)

commit d39d358c (2017-05-15): input-raw: fix multitouch devices without BTN_TOUCH

For type B devices, the tracking id is enough and there are devices out there
that rely on that.

Fixes #97