libts/tslib #98: FR: Version numbers in changelog

jonasstein, 2017-05-16:
It would be nice to have a version number behind the date in the changelog.
This would help package maintainers. Thank you.

merge, 2017-05-16:
We have the changelog in the NEWS file, https://github.com/kergoth/tslib/blob/master/NEWS and copy it to the "release pages" that host the tarballs:
* https://github.com/kergoth/tslib/releases and
* https://gitlab.com/tslib/tslib/tags (mirror)

The version number is the header. Do you mean, to have a date behind the version number?

jonasstein, 2017-05-16:
I just looked in the changelog file where its 

2006-08-16 Chris Larson <clarson@kergoth.com>
	* plugins/collie.c,plugins/corgi.c: Apply patch from Richard Purdie to
	  fix the microseconds field of the timeval struct, as it was
inadvertantly holding milliseconds.

Do I understand it correct, that there are two files for changes? news and changelog, which could be merged?

merge, 2017-05-16:
Yes, you are right. I know this is misleading. The ChangeLog file is still around, but had been dismissed since the project had used git.

I maintain the NEWS file to manage library releases and collect the release changelog for users.

There had been an idea of reworking the git history and incorporating all information of the ChangeLog file into the git history, and then delete it, but I doubt that this will be worth it.

We'll have to merge them somehow. I'm not yet certain how exactly.

jonasstein, 2017-05-16:
I see. For me it would look reasonable to merge just the two text files and create a uniform format. And then just delete the obsolete changelog. This would be not too much work and the confusion is gone. Just a suggestion...

commit 5ac918e0 (2017-05-17): archive ChangeLog file by moving it to doc/ChangeLog.old and point to it

Fixes #98
closed 2017-05-17 by merge