libts/tslib #99: input-raw.c:198: error: 'SYN_MT_REPORT' undeclared (first use in this function)

bujiebuhuo, 2017-05-18:
When I compile tslib-1.4 I met that question, and I find there isn't "SYN_MT_REPORT" in tslib, I want to know how to solve it. 
 
input-raw.c: In function 'check_fd':
input-raw.c:198: error: 'SYN_MT_REPORT' undeclared (first use in this function)
input-raw.c:198: error: (Each undeclared identifier is reported only once
input-raw.c:198: error: for each function it appears in.)
input-raw.c: In function 'ts_input_read_mt':
input-raw.c:484: error: 'SYN_MT_REPORT' undeclared (first use in this function)
make[2]: *** [input-raw.lo] Error 1
make[2]: Leaving directory `/mnt/hgfs/Download/tslib-1.4/plugins'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/mnt/hgfs/Download/tslib-1.4'
make: *** [all] Error 2

merge, 2017-05-18:
use tslib 1.10 please

bujiebuhuo, 2017-05-18:
Thank you，I am cross-compile, tools/ts_uinput.c also use SYN_MT_REPORT,I also meet this error.

merge, 2017-05-18:
those are the 2 linux dependent files. If you don't need ts_uinput you can disable it with configure --disable-tools.

If you have the problem with 1.10, you have a few options:
* upgrade the kernel
* define it in input-raw.c and send a patch

merge, 2017-05-18:
actually this should be fixed in tslib 1.10. Please use it.

bujiebuhuo, 2017-05-19:
I find the define of "SYN_MT_REPORT" in linux kernel file.It should be I have't configure cross-compiler well.Thank you so much.

bujiebuhuo, 2017-05-19:
I am now in the cross compiler, I would like to know how to put my embedded linux kernel file directory added. I have been using cmake + make, not familiar with autotools.

merge, 2017-05-19:
what kernel and tslib versions are you trying to cross compile?

bujiebuhuo, 2017-05-19:
linux 3.0.1 and tslib1.10

merge, 2017-05-19:
First of all, you have to upgrade your kernel. See kernel.org for supported version. Seriously. Do it. No matter how "not connected" your device may be.

But anyways, at least building with `./configure --disable-tools` should work; because I'm not sure whether ts_uinput runs on such an old kernel.

cross compiling tslib is just like cross compiling any autotools project, so you might want to read up on that in general.

bujiebuhuo, 2017-05-19:
Thank you very much for your help, my problem is glic is too old.

bujiebuhuo, 2017-05-19:
Thank you very much for your help, my problem lies in the cross compiler glibc is too old.

merge, 2017-05-19:
Ok. I'm not sure if we can improve tslib's build config here because glibc is not the only C library we support.

merge, 2017-05-19:
If I get it right, you may close this issue. Otherwise may I will do so soon, in order to clean up. But feel free to open one in case you have any other problem or question regarding tslib. Of course you can also always write to tslib@lists.infradead.org

thanks

closed 2017-05-19 by merge