libts/tslib #100: Is there a way to generate trackid for a multi-touch device without trackid

lizheng851123, 2017-06-03:
i am using a multi-touch pannel that do not report trackid, the drvier just repot the samples in  multitouch protocolA without ABS_MT_TRACKING_ID  . 
is three a way that tsllib can generate trackid for a device like that?
maybe a new plugin?

merge, 2017-06-04:
libmtdev does that: http://bitmath.org/code/mtdev/

tslib basically already generates slots for you and generating a tracking id would be possible too... But for touch-start / end tslib provides the pressure value. It's generated if the device doesn't have pressure.

why do you need this? What do you want to do?

merge, 2017-06-04:
That said, I think we should add this by default and generate valid tracking id values if the device doesn't have them. Just like we do for pressure.

I currently dont have a type A device but I'll work on it anyways, probably for 1.12.

Thanks for bringing this up!

merge, 2017-06-08:
protocol type A is deprecated and all kernel drivers are converted to type B now. For old kernel, I guess I'll still add something here. But a __very__ simple logic (probably just counting up an integer with pen-down until it overflows) that is compatible with the spec. This should be better than nothing. And I'll do it in the input-raw plugin.

commit d12a63f6 (2017-06-29): input-raw: insert a valid tracking id for type A devices

Generate a tracking id value for devices that don't provide one. We simply
increment until we overflow.

Fixes #100
closed 2017-06-29 by merge