libts/tslib #101: ./configure -h problem

lzwgg, 2017-06-22:
using the tslib 1.11
after I used ./autogen.sh, I typed the ./configure -h, and encountered this in optional features:
`
Optional Features:
  --disable-option-checking  ignore unrecognized --enable/--with options
  --disable-FEATURE       do not include FEATURE (same as --enable-FEATURE=no)
  --enable-FEATURE[=ARG]  include FEATURE [ARG=yes]
  --enable-silent-rules   less verbose build output (undo: "make V=1")
  --disable-silent-rules  verbose build output (undo: "make V=0")
  --enable-dependency-tracking
                          do not reject slow dependency extractors
  --disable-dependency-tracking
                          speeds up one-time build
  --enable-static[=PKGS]  build static libraries [default=no]
  --enable-shared[=PKGS]  build shared libraries [default=yes]
  --enable-fast-install[=PKGS]
                          optimize for fast installation [default=yes]
  --disable-libtool-lock  avoid locking (might break parallel builds)
  --enable-linear         Enable building of linear scaling (default=yes)
  --enable-dejitter       Enable building of dejitter filter (default=yes)
  --enable-linear-h2200   Enable building of linearizing filter for iPAQ h2200
                          (default=yes)
  --enable-variance       Enable building of variance filter (default=yes)
  --enable-median         Enable building of median filter (default=yes)
  --enable-pthres         Enable building of pthres filter (default=yes)
  --enable-debounce       Enable building of debounce filter (default=yes)
  --enable-skip           Enable building of skip filter (default=yes)
  --enable-iir            Enable building of IIR filter (default=yes)
  --enable-ucb1x00        Enable building of ucb1x00 raw module (UCB1x00
                          support) (default=yes)
  --enable-cy8mrln-palmpre
                          Enable building of cy8mrln-palmpre raw module
                          (default=no)
  --enable-corgi          Enable building of corgi raw module (Sharp Zaurus
                          sl-c7x0 support) (default=yes)
  --enable-collie         Enable building of collie raw module (Sharp Zaurus
                          sl-5500/5000d support) (default=yes)
  --enable-h3600          Enable building of h3600 raw module (HP iPaq H3600
                          support) (default=yes)
  --enable-mk712          Enable building of mk712 raw module (Hi tachi
                          support) (default=yes)
  --enable-arctic2        Enable building of arctic2 raw module (IBM Arctic II
                          support) (default=yes)
  --enable-tatung         Enable building of tatung raw module (Tatung Webpad
                          support) (default=yes)
  --enable-dmc            Enable building of dmc raw module (HP iPaq DMC
                          support) (default=yes)
  --enable-dmc_dus3000    Enable building of dmc_dus3000 raw module (DMC DUS
                          Series support) (default=no)
  --enable-input          Enable building of generic input raw module (Linux
                          /dev/input/eventN support) (default=yes)
  --enable-touchkit       Enable building of serial TouchKit raw module (Linux
                          /dev/ttySX support) (default=yes)
  --enable-galax          Enable building of HID USB eGalax raw module (Linux
                          /dev/hiddevN support) (default=yes)
  --enable-waveshare      Enable building of WaveShare raw module (Linux
                          /dev/hidrawN support) (default=yes)
  --enable-waveshare      Enable building of WaveShare raw module (Linux
                          /dev/hidrawN support) (default=no)
  --enable-input          Enable building of generic input raw module (Linux
                          /dev/input/eventN support) (default=no)
  --enable-galax          Enable building of HID USB eGalax raw module (Linux
                          /dev/hiddevN support) (default=no)
  --enable-touchkit       Enable building of serial TouchKit raw module (Linux
                          /dev/ttySX support) (default=no)
  --enable-ucb1x00        Enable building of ucb1x00 raw module (UCB1x00
                          support) (default=no)
  --enable-corgi          Enable building of corgi raw module (Sharp Zaurus
                          sl-c7x0 support) (default=no)
  --enable-collie         Enable building of collie raw module (Sharp Zaurus
                          sl-5500/5000d support) (default=no)
  --enable-h3600          Enable building of h3600 raw module (HP iPaq H3600
                          support) (default=no)
  --enable-mk712          Enable building of mk712 raw module (Hi tachi
                          support) (default=no)
  --enable-arctic2        Enable building of arctic2 raw module (IBM Arctic II
                          support) (default=no)
  --enable-tatung         Enable building of tatung raw module (Tatung Webpad
                          support) (default=no)
  --enable-dmc            Enable building of dmc raw module (HP iPaq DMC
                          support) (default=no)
  --enable-debug
		Enable debug messages from filters (default=no)

  --disable-tools         Disable tools (ts_uinput)
`
I found there are several options which defined "yes" and "no" together.
Are they correct?
Hoping your reply. Thanks.

merge, 2017-06-22:
hm. it seems like host_os isn't uniquely found in your case. What operating system do you run it on?

lzwgg, 2017-06-22:
I am using Ubuntu 14.04 in VMware workstation.

commit 74f527e6 (2017-06-28): configure.ac: fix help text by defining only one default configuration

And document what modules can be enabled, or have to be disabled, for building
on various platforms.

The vast majority of users are on Linux, and those are unaffected by this
change. This fixes a macro issue we have had with configure's help text.

Fixes #101
closed 2017-06-28 by merge