libts/tslib #102: Not all touchscreens provide EV_KEY

craftyguy, 2017-09-03:
I'm writing a touchscreen keyboard for phones, and have come across a number of devices (mainly Android) which have touchscreen devices without EV_KEY support. As you may expect, this causes tslib to bail out immediately with a message that "this is not a touchscreen device". 

[The reason EV_KEY isn't present in the drivers for these devices seems to be because it was not needed for Android.](https://stackoverflow.com/questions/33974570/why-didnt-i-receive-the-ev-key-event-on-some-rooted-android-devices-when-requ)

```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	barometer
/dev/input/event1:	Melfas MMSxxx Touchscreen
/dev/input/event2:	tuna-gpio-keypad
/dev/input/event3:	proximity
/dev/input/event4:	lightsensor-level
/dev/input/event5:	Tuna Headset Jack
Select the device event number [0-5]: 1
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "Melfas MMSxxx Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 3 (EV_ABS)
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max       30
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      719
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     1279
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
    Event code 58 (ABS_MT_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
```

merge, 2017-09-03:
Thanks for reporting! What version of tslib are you using here?

ollieparanoid, 2017-09-03:
Thanks for the quick answer @merge! I'm also involved with the project @craftyguy is working on, so I looked up the version. It is [1.12](https://github.com/postmarketOS/pmbootstrap/commit/07dccafe42f53ea2b04db6471113cb03825f5517).

merge, 2017-09-03:
True. I see the bug. It'll be fixed for 1.13. I'll still test if we can check BTN_TOUCH unconditionally or not. thanks again!

craftyguy, 2017-09-03:
Thank you for the quick response and action! We really appreciate it! If you need us to test anything before you cut the 1.13 release, please let us know!

merge, 2017-09-04:
There is a fix in master. Can you test that? In any case I guess I'll release a 1.13-rc1 not too long from now, and cut this cycle a little short.

closed 2017-09-04 by merge

Pneumaticat, 2017-09-04:
As a tester with one of these multitouch-only phones, I can confirm that master fixes the problem! The touchscreen works correctly in `ts_test_mt`.

However, it gives no output in `ts_test`, seemingly because `ts_test` uses the singletouch-only `ts_read` function. Since the touchscreen only emits multitouch events, is it intended behavior that the singletouch API gives no output? (It makes sense, but I'm not sure and it breaks some applications that only use `ts_read`, like DirectFB, so I just wanted to confirm it.)

craftyguy, 2017-09-05:
@merge 

Thank you again, this seems to be working great!

merge, 2017-09-05:
If anyone of you 3 currently writing in this thread give me a name and email address, I'd add you to the THANKS file.

so, thanks so far.

craftyguy, 2017-09-05:
Sure!

Clayton Craft 
clayton -at- craftyguy -dot- net

Pneumaticat, 2017-09-05:
Sure!

Kevin Liu <kevin@potatofrom.space>

ollieparanoid, 2017-09-05:
(No need to add me, I just looked up a version.)
It's a pleasure to work with such friendly upstream people like you @merge 👍

commit b8594cca (2017-09-04): input-raw: Don't require to have the EV_KEY event type

We already didn't require to have BTN_TOUCH or BTN_LEFT because
multitouch type A and B can do without them. And since these are the
only EV_KEY event code we check, we obviously can remove the requirement
for the EV_KEY type.

Fixes #102