libts/tslib #104: ts_read does not work for multitouch-only device without EV_KEY

Pneumaticat, 2017-09-05:
Hi,

Thanks to b8594cca450ee2b07476f0cbe5fb806c13289ddc, my touchscreen (a Melfas MMSxx on a Galaxy Nexus) now works with tslib despite not emitting the singletouch API's EV_KEY event (as detailed in #102). Now, `ts_test_mt` works, as do programs using the `ts_read_mt` API. However, `ts_test` and programs (like DirectFB) using `ts_read` do not, though from #103 it seems that both `ts_read` and `ts_read_mt` should work for multitouch devices.

Perhaps the problem is that the touchscreen _only_ implements the Linux multitouch API, and not the single-touch one? Here's an `evtest` snippet from a single touch:

```
Event: time 115.790557, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 115.790588, type 3 (EV_ABS), code 48 (ABS_MT_TOUCH_MAJOR), value 13
Event: time 115.790588, type 3 (EV_ABS), code 58 (ABS_MT_PRESSURE), value 183
Event: time 115.790588, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 345
Event: time 115.790588, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 371
Event: time 115.790618, -------------- SYN_REPORT ------------
Event: time 115.808166, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value -1
Event: time 115.808166, -------------- SYN_REPORT ------------
```

Here's the `evtest` info, copied from #102:

```
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:	barometer
/dev/input/event1:	Melfas MMSxxx Touchscreen
/dev/input/event2:	tuna-gpio-keypad
/dev/input/event3:	proximity
/dev/input/event4:	lightsensor-level
/dev/input/event5:	Tuna Headset Jack
Select the device event number [0-5]: 1
Input driver version is 1.0.1
Input device ID: bus 0x18 vendor 0x0 product 0x0 version 0x0
Input device name: "Melfas MMSxxx Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 3 (EV_ABS)
    Event code 47 (ABS_MT_SLOT)
      Value      0
      Min        0
      Max        9
    Event code 48 (ABS_MT_TOUCH_MAJOR)
      Value      0
      Min        0
      Max       30
    Event code 53 (ABS_MT_POSITION_X)
      Value      0
      Min        0
      Max      719
    Event code 54 (ABS_MT_POSITION_Y)
      Value      0
      Min        0
      Max     1279
    Event code 57 (ABS_MT_TRACKING_ID)
      Value      0
      Min        0
      Max    65535
    Event code 58 (ABS_MT_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
  Property type 1 (INPUT_PROP_DIRECT)
```

merge, 2017-09-05:
Thanks for reporting. I'll have a look during the upcoming days...

merge, 2017-09-05:
Has it worked before? Which version of tslib works, which doesn't?

Pneumaticat, 2017-09-05:
I don't think it has ever worked in the past -- this touchscreen only started working at _all_ with b8594cca450ee2b07476f0cbe5fb806c13289ddc, which was only a few commits ago. Previously there was a check for EV_KEY (which seems to have been around for at least 9 years) that prevented it from working.

craftyguy, 2017-09-05:
Here is the `ts.conf` we are using: https://ptpb.pw/mkAO

merge, 2017-09-05:
Thanks for the data. Up until now this has been intended behaviour. While ts_read_mt() is the far better choice, also from the filters' point of view, I'll think about whether supporting ts_read() is worth it. In the end we'd have to rethink differentiating between MT and singletouch devices at all.

That said I'm happy to help supporting ts_read_mt() in your application too.

merge, 2017-09-06:
It might be worth testing ts_read() (or directfb, whatever) with this device using the current git code here again.

craftyguy, 2017-09-06:
Thanks, building a package now for users to test this on MT-only devices

Pneumaticat, 2017-09-07:
Thanks for the quick response, @merge!

9f2345af5f93b69731bff6e61681e80d2a55a914 partially fixes the issue (so that `ts_test` can track movement with my MT-only touchscreen), but it still seems to not register releases (that is, it never outputs an event of pressure=0 when lifting my finger).

Here's a run of ts_test where I repeatedly tapped the screen:

```
67.145935:    400    835    172
67.221160:    400    835    172
67.799804:    384    857    172
67.865051:    380    863    172
68.347412:    382    884    174
68.422515:    382    901    176
68.906585:    379    906    173
68.961090:    379    912    171
69.250854:    381    886    188
69.269958:    381    869    197
69.289031:    384    858    189
69.298217:    387    847    161
69.307189:    387    848    135
69.577575:    391    850    157
69.605895:    393    852    132
69.614318:    394    853    126
```

None of the events show pressure=0 in the third column.

merge, 2017-09-08:
thanks for the update, I thought so. I pushed another fix on top of it now. Please note that, although I think it should work and closed the issue in this commit, I didn't test, so please re-open it, if it doesn't yet work for you!

thanks for testing!!

closed 2017-09-08 by merge

craftyguy, 2017-09-08:
@merge 

Thank you so much! I'll get this out to folks to test ASAP. We REALLY appreciate your hard work and willingness to help :)

Pneumaticat, 2017-09-08:
Confirming that it works! Touches are now detected by DirectFB. Thanks so much for your help!

merge, 2017-09-09:
my pleasure. thanks for testing so quickly.

commit 1ad02d41 (2017-09-08): input-raw.c: support MT-only devices in ts_read()

struct ts_sample is pressure-based. MT-only devices don't even have to
report an EV_KEY type like BTN_TOUCH. Their pen_up is tracking_id == -1.

Fixes #104