libts/tslib #105: To all users: Do you use ANY other module_raw than "input" in your ts.conf?

merge, 2017-09-06:
We certainly have useless `input_raw` modules mentioned in ts.conf and enabled by default `./configure` on Linux. The question is: Which ones are definitely useless enough that people who **do** use them, have them explicitely enabled?

We'll always have `module_raw input` configured in by default in Linux. This is what people **should** be using. This is the **only** module that supports multitouch! Other than that, we have the following candidates I would like to evaluate:

* arctic2
* collie
* corgi
* dmc
* galax (deprecated. can be safely replaced by `input`)
* h3600
* mk712
* tatung
* touchkit
* ucb1x00
* waveshare

### If you use any module_raw from this list in your ts.conf, please reply here and say so!

If you don't (and in any case) make sure you enable it explicitely via `./configure --enable-arctic2` and so on.

craftyguy, 2017-09-06:
My application targets Linux, and so I just use the default `module_raw input` since we have not come across any devices that require otherwise to function.

merge, 2017-09-06:
The low hanging fruits would probably the Linux-specific ones, galax and waveshare. I'll consider disabling them quite soon.

The others are usable on Mac OS and Hurd too.

ClayMontgomery, 2017-09-30:
My application only uses module_raw input.

merge, 2018-09-13:
Alright this is running for long enough and we seem to have identified what we want to keep enabled by default. thanks.

closed 2018-09-13 by merge