libts/tslib #109: Required tslib utils don't get installed on Yocto linux builds

ClayMontgomery, 2017-09-30:
I reported a new install bug regarding tslib for the Yocto project and I will link it here also:

https://bugzilla.yoctoproject.org/show_bug.cgi?id=12147

The problem is that these utils are not getting installed into the root filesystem for embedded linux builds using the newer versions of the Yocto project named Pyro (2.3).

ts_calibrate
ts_finddev
ts_harvest
ts_print
ts_print_mt
ts_print_raw
ts_test
ts_test_mt
ts_uinput

Of course, tslib does not function at all until ts_calibrate has been run at least once on a new linux system to create the pointercal file. This issue appears to be in the recipes for Yocto Pyro.

merge, 2017-10-01:
Actually it's only the "linear" plugin that need ts_calibrate :)

installing these programs can't any different than with any other project... There's nothing special about tslib.

merge, 2017-10-02:
Thanks for bringing the issue up and keeping us informed! As I don't see any problem with tslib here, I'll close the issue. An email would probably have been enough :)

I hope yocto or the respective OE repo can upgrade to tslib 1.13.

closed 2017-10-02 by merge