libts/tslib #110: Yocto project needs to update from tslib version 1.1

ClayMontgomery, 2017-09-30:
Yocto is becoming the most widely used environment for building embedded linux systems. But, they are still stuck on tslib 1.1, which does not even support multi-touch.

 I think we should request that the Yocto Project update. Are there any known issues with such a pull request? Which version of tslib would be appropriate?

merge, 2017-10-01:
I tried once but didnt bother registering to their lists. I don't use yocto atm. What stops you from doing it? The newest version of course. The API is backwards compatible. The soname changed to "tslib.so.0" but that shouldn't be an issue for a source distro.

merge, 2017-10-02:
I have the feeling that this bug should have been file over at https://bugzilla.yoctoproject.org/ so I'll close this issue. Feel free to reply either way though.

closed 2017-10-02 by merge