libts/tslib #111: out-of-source build broken

DerDakon, 2017-10-19:
When I try to build out of source like this:

```shell
mkdir build
cd build
../tslib/configure
````
The build breaks with errors like this:

```
make[2]: Entering directory '/tmp/build/src'
Makefile:622: ../../tslib/plugins/.deps/arctic2-raw.Plo: No such file or directory
Makefile:623: ../../tslib/plugins/.deps/collie-raw.Plo: No such file or directory
```

Looks like $(top_srcdir) isn't properly expanded then. You can also see this:

```shell
$ ls -1 src/
Makefile
$(top_srcdir)
```

The last line is no typo, the directory actually is named that way.

Happens with current git HEAD as well as 1.13, both for native and cross builds.

merge, 2017-10-20:
Thanks for reporting this! try `..tslib/configure --disable-dependency-tracking` which should work.

I'm not yet sure what we are doing wrong here and I'm happy for hints. It looks like this https://lists.gnu.org/archive/html/bug-automake/2004-01/msg00044.html or https://lists.gnu.org/archive/html/bug-automake/2014-01/msg00005.html but it'd be strange if that wouldn't be fixed by now.

disabling subdir-objects isn't really an option, as this will be enabled by default in future automake versions and it's not appropriate for use anyways.

Related to this problem might be that `make distclean` also only works when using `--disable-dependency-tracking`.

Could we force `--disable-dependency-tracking` by default? It doesn't seem like the right solution, but what would we lose? If we don't have a proper solution soon, I think about trying to force this for the next release. I have ignored this for too long now, so thanks a lot for bringing it up :) It helps to solve this once and for all.

merge, 2017-10-20:
`--disable-dependency-tracking` is now the default. You can still use `--enable-dependency-tracking` to turn it on. This should fix the following:

* out-of-tree building
* make distclean
* make distcheck

closed 2017-10-20 by merge

merge, 2017-10-20:
@DerDakon If this doesn't fix the issue for you, please feel free to re-open this. And either way, if you give me your name and email, I'd like to add you to the THANKS file, if you want that. so: thanks.

commit 93ac81dc (2017-10-20): autoconf: disable dependency tracking by default

Since make distclean and out-of-source builds don't work when using
the (enabled by default) dependency tracking, let's disable it by
default. This also fixes make distcheck.

This describes our problem: https://stackoverflow.com/q/21609580/6435680

If this is not a bug in autotools, we'll need a real solution here. But
if we don't refactor much, we don't lose much during development. Plus,
we can simply use --enable-dependency-tracking.

Closes #111