libts/tslib #113: ts_uinput: do we have to take x/y max values from the framebuffer device?

merge, 2017-11-02:
Point is, I want to make the TSLIB_FBDEVICE and TSLIB_CONSOLEDEVICE environment variables deprecated, at least as soon as `./configure --with-sdl2` is used. ts_uinput uses the framebuffer device for getting x/y maximum values though. What happens if we assume a touchscreen's input device offers the same values?

craftyguy, 2017-11-02:
Is there a way to test/check the coordinate system for tslib using framebuffer vs tslib getting coordinates directly from touch devices?

merge, 2017-11-02:
tslib (or the library) itself doesn't really use any coordinate system. It takes x/y values as they come. The device drivers I have seen so far use the same coordinate system for both. Technically it doesn't matter, especially when you apply the `linear` filter (ts_calibrate) anyways. In this case ts_uinput is the device driver (userspace input device driver) that reads the resolution from the framebuffer device.

Strictly speaking we must not break ts_uinput behaviour and, if anything at all, write a different one, that uses the input device driver only.

Only in case we could verify that it simply always is the same on Linux, we could change ts_uinput. That's the question.

There's no simple switch from one to the other in ts_uinput, but we could add an option... That's probably the first step here.

craftyguy, 2017-11-02:
Oh ok. Maybe I misunderstood your original question. It seems that if we're using `linear` that there would be no change to behavior regardless of whether you're reading resolution from fbdev or directly from the touch device?

merge, 2017-11-03:
Absolutely. If you change resolution values, you'd need to re-calibrate though.

commit e06e4430 (2017-11-30): ts_uinput: add --nofb option that doesn't access the fb device

It simply uses the input device information to get the "screen"
besolution.

Closes #113
closed 2017-11-30 by merge