libts/tslib #114: Flickering cross in center of ts_test_mt

ollieparanoid, 2017-11-09:
When I move 3 fingers across the screen, it shows 3 cursors at the right locations (but also one flickering cursor in the middle when not moving the fingers).

@merge wrote in https://github.com/kergoth/tslib/issues/108#issuecomment-343063113:
> Yeah the flickering cross in the center of ts_test_mt... It shouldn't be there, it's ugly and a bug basically. On the other hand, it indicates that there is an unused available slot still available, which I personally find handy :)
> 
> But if you want to, create and issue for that. If we want to indicate how many slots are available in total, we should do so explicitely. We could write a small number to a corner of the screen?

I have only used `ts_text_mt` to quickly test one awesome feature @merge has implemented, so I'm not affected at all by this. But I think it is good practice to keep bugs in the tracker, so here's the issue about it.

Writing a small number on the screen sounds like a good solution to me 👍

closed 2017-11-13 by merge

merge, 2017-11-14:
So the flickering should be fixed. And I now write "X touch contacts supported (<reason>)" where reason is either "driver" or "user":

[image]

now, it this too much text at the bottom? It it self-explanatory? Thanks @ollieparanoid

ollieparanoid, 2017-11-14:
I think it's perfect 👍

commit 47fb6162 (2017-11-13): ts_test_mt: avoid a flickering cross in the center

Fixes #114