libts/tslib #116: doesn't work on new versions of Android (x86)?

shalnoff, 2017-11-15:
I've been trying to calibrate x86 Android 7.2 version running on VM, and it doesn't work anymore.
Early versions has been calibrated very well (following [this instruction](https://androidcarx86.blogspot.de/2012/04/android-x86-and-touch-screen.html))

Is there any way to use it on newer versions of Android?

merge, 2017-11-15:
What exactly doesn't work? Do you have tslib installed?

The instructions you linked to should work, and in case there is a problem with tslib on Android, I'm always happy to take a patch or pull request.

Also, in case you have Android x86 and build tslib yourself, could you show me how to build or get that VM image, and how you do it? thanks

merge, 2017-11-17:
I verified that tslib (1.15 will again) build on Android.

The thing is, while we maintain an "Android.mk" file to be up-to-date inside of our source tree, we don't further support any Android streamlining. (I don't use tslib on Android so far)

But all you need besides Android.mk is an appropriate "config.h" file that is generated by "./configure", see for example https://stackoverflow.com/a/10356958/6435680

I don't really think that we can do more than maintaining an "Android.mk" file, but I'd be very happy for a nice documentation how to include tslib in AOSP. I don't even know how to properly add it to a build/target/product/... file :)

So, I don't think we make it hard for you to include tslib in Android. My guess would be that you don't have tslib included in your system, so I'll go ahead and close this, as it's not at all our business. But please re-open the issue if you *have* tslib but have problems using it!! And if you can, please write some documentation on how to properly include tslib in AOSP.

thanks

closed 2017-11-17 by merge