libts/tslib #117: Improve support for Android

merge, 2017-11-17:
I know very little about Android system developemnt! This all is not set in stone, but are merely thoughts that came up after building tslib once inside of AOSP:

tslib builds file in AOSP, given you have an appropriate `config.h` file. Therefore the configure scripts needs to be run, with the Android build environment variables set.

I see 3 options how we can improve things:

* include our Android.mk in the release tarball and document how to conveniently include that as `external/tslib/` AND document a convenient way of running the configure script in order to get the needed `config.h` file.

* document how to conveniently run autoreconf in order to obtain the configure script that in turn generates the `config.h` file inside of AOSP `external/tslib`

* remove the Android.mk file and create a seperate android_external_tslib project, that "depends" on tslib and pulls it in?

How do other autotools based projects that are included in AOSP do this?

and optionally, and actually unrelated in a first step:

* create a new project, "ts_calibrate" as an Android app, releasing APK packages (plus documentation how to include it in AOSP).

licheegh, 2018-01-18:
Hi~I am doing the same thing now.
My plan is:
1. build tslib in android.
2. use ts_uinput to create a input event for android system to use.
3. create a app on android to generate calibrate data.
I don't know whether it will work or not. But I will try.

merge, 2018-01-18:
Interesting. Do you develop it in public somewhere? I'm happy for any feedback.

How do you go about building tslib in Android? Do you have a proper way to run `./configure` for Android?

licheegh, 2018-01-19:
I just start it and I don't write any code yet, I will post here if I have a update.

I am using this [android BSP](http://www.phytec.de/documents/l-846e-0-imx6-android-bsp-manual/), well it is commercial, but helps to understand the enviroment I am using I think.

I git clone tslib in the path `externel/tslib`, then source the build enviroment which is used to build android system.

```bash
./autogen.sh
./configure
mm
```

then build the android system.

after that I can have have .so files in rootfs and also ts_calibrate can be run in adb. But right now various tslib commands will print segment fault when calling `ts_close()`. I will try to fix this and test `ts_uinput`

merge, 2018-01-19:
Remember to always include your ts.conf file, i.e. the modules you use, when reporting bugs. And also for you when you run into a bug: Disable all modules and try again. Then the bug should be in the core, otherwise in a module...

licheegh, 2018-01-30:
We have successfully using 'ts_uinput' in Android~

And have two major issues solved:

## No event from uinput device

We are using master version tslib, under linux it works fine. But under Android the eventX created by ts_uinput do not send out any event. After debug we find that the uinput interface under our Android's kernel can only being write once per event. (its kernel version is v3.10.53)

So we create this diff

```diff
diff --git a/tools/ts_uinput.c b/tools/ts_uinput.c
index e2bfb67..ccd1323 100644
--- a/tools/ts_uinput.c
+++ b/tools/ts_uinput.c
@@ -161,6 +161,8 @@ static int send_touch_events(struct data_t *data, struct ts_sample_mt **s,
 	int i;
 	int j;
 	int c = 0;
+	int ret = 0;
+	int k = 0;
 
 	for (j = 0; j < nr; j++) {
 		memset(data->ev,
@@ -318,14 +320,24 @@ static int send_touch_events(struct data_t *data, struct ts_sample_mt **s,
 			data->ev[c].type = EV_SYN;
 			data->ev[c].code = SYN_REPORT;
 			data->ev[c].value = 0;
-
-
+#if 0
 			if (write(data->fd_uinput,
 				  data->ev,
 				  sizeof(struct input_event) * (c + 1)) < 0) {
 				perror("write");
 				return errno;
 			}
+
+#else
+			for(k = 0; k <=c; k++) {
+				ret = write(data->fd_uinput, &data->ev[k], sizeof(struct input_event));
+				if (ret < 0) {
+					perror("write");
+					return ret;
+				}
+				
+			}
+#endif
 		}
 
 		c = 0;
```

which is write to uinput every event. Then we can see the events in getevent under Android

## Two input device activate simultaneously

Android will recognize two input device (one is touch screen event, other is from ts_uinput), we need to disable touch screen one using

`excluded-input-devices.xml`

like
https://github.com/pakesson/ParallelDroid/blob/master/android/system/etc/excluded-input-devices.xml
https://github.com/maruos/vendor_maruos/blob/master/excluded-input-devices.xml

as following

```xml
<devices>
    <device name="stmpe-ts" />
</devices>
```

yeah, we are using stmpe811. Then we can start ts_uinput from .rc file.

merge, 2018-01-30:
Thanks for reporting all that! I'll look into whether we can do your change in ts_uinput.

Could you also show your exact steps to *configure* and build tslib inside of external? We could find a place to document that.

merge, 2018-02-13:
Thanks again, your change to ts_uinput doesn't seem to be a problem and is now applied and will be part of a next release.