libts/tslib #118: Option to invert x and/or y axe

hevilp, 2017-11-28:
Hello,

I'm using Libreelec, which is using tslib. I got the problem, that the x axe is inverted.
Is there any posssibility to make a module to get the option to invert x and/or y axe?

merge, 2017-11-28:
Hi,

There is no explicite "invert values" filter. But that'd be quite tricky to do. What exactly would "invert" be? Just multiplied -1? mirrored around some value? We could work it out however.

BUT: Did you use the ts_calibrate program and have `module linear` in your ts.conf? Please make sure that you run ts_calibrate again, verify the calibration file (probably called "pointercal") is generated. In case you have done this and still have problems using your touchscreen, we definitely have to think about adding a filter. If I understand correctly, this should fix your problem.

That said, please describe the "inverted" behaviour in a little more detail. In case others have the same problem, we can think about adding a filter for that, that doesn't require manual calibration and is faster.

merge, 2017-11-28:
Ok. I'm sorry, but I don't yet understand how you "can't calibrate". As long as you __see__ the cross on the screen, you __can__ calibrate. Just touch it. Why would you need a "cursor"? I'm really sorry. Please make me understand it! I really want to solve this :) thanks for reporting the problem!

merge, 2017-11-28:
Would you paste the output of `evtest` when you move from the very top of your screen, to the very bottom. You have to select your _original_ touchscreen device with `evtest`. Can you do that?

thanks.

hevilp, 2017-11-28:
I will do it later this evening (germany)

Kind regards

hevilp, 2017-11-29:
Hi,

I dont know why, but I got 2 events for 1 screen:

`No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      eGalax Inc. Touch
/dev/input/event1:      eGalax Inc. Touch
/dev/input/event2:      Logitech USB-PS/2 Optical Mouse
/dev/input/event3:      tslib to uinput daemon
`
When I run evtest it comes:

`  This device is grabbed by another process.
  No events are available to evtest while the
  other grab is active.
  In most cases, this is caused by an X driver,
  try VT-switching and re-run evtest again.
  Run the following command to see processes with
  an open fd on this device
 "fuser -v /dev/input/event1"
`
So I kill with `fuser -v /dev/input/event0` or `event1`.
If I do that, Kodi restarts (using libreelec)
After Kodi restarts automatically I run evtest and it comes:
`Select the device event number [0-2]: 0
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xeef product 0x1 version 0x112
Input device name: "eGalax Inc. Touch"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min       42
      Max     1981
    Event code 1 (ABS_Y)
      Value      0
      Min       42
      Max     1981
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
`
I cannot touch the crosshair. The crosshair seems to be outside of the toucharea.

hevilp, 2017-11-29:
[image]

merge, 2017-11-29:
Can you still move your finger from top to bottom and record what evtest prints? Independently of any running program. Just switch to a new virtual terminal for example.

Also. You (or kodi) don't use tslib's `ts_calibrate` program for calibration. Unfortunately I don't know the program you are using. I can only assume it behaves the same, but I doubt it. I don't know why it doesn't show the cross... So, after having your evtest recording for me, you could try to run ts_calibrate, if you have it.

hevilp, 2017-11-29:
Are you sure that Kodi doesnt use ts_lib? I'm pretty sure, that it does.

I stop Kodi, kill all processes that using event0 and run evtest with it.

I get in the console `
Select the device event number [0-2]: 0
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0xeef product 0x1 version 0x112
Input device name: "eGalax Inc. Touch"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 272 (BTN_LEFT)
    Event code 273 (BTN_RIGHT)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min       42
      Max     1981
    Event code 1 (ABS_Y)
      Value      0
      Min       42
      Max     1981
  Event type 4 (EV_MSC)
    Event code 4 (MSC_SCAN)
Properties:
Testing ... (interrupt to exit)
` but nothing happens when I touch it.

hevilp, 2017-11-29:
Okay, here is the log
[log.txt](https://github.com/kergoth/tslib/files/1512873/log.txt)

merge, 2017-11-29:
It does use (an old version of) tslib, or more accurately, the libts library. But it doesn't use the calibration program that comes with tslib. The program you are using uses the libts library, which is fine. I only don't know it.

Wait. `evtest` __must__ print values when you select the original touchscreen event file. If it doesn't, tslib (anything using your touchscreen) will never be able to work. That's for sure, because there simple __are__ no input samples coming from your device if evtest doesn't print anything. So first make sure your touch panel isn't broken or your Linux driver is working until evtest prints samples while touching.

If your cursor is moving (inverted x-axis as you said), then you might have a setup problem or there is still something running that uses the touchscreen input device. That's also clear, because otherwise __nothing__ could move a cursor :)

hevilp, 2017-11-29:
Watch my answer above yours. There is the log with my finger moved from top to down

hevilp, 2017-11-29:
`ts_calibrate
xres = 1, yres = 1
tslib: Unable to grab selected input device`

I am unable to run ts_calibrate.

merge, 2017-11-29:
Alright. So your y-axis is inverted. Over at libreelec's patched version of tslib there is a "st1232" module_raw with a "invert_y" parameter you could try to use. Try that!

I don't really get why it's inverted though and who might be to blame here. But I don't know Kodi or Libreelec well at all.

As long as Libreelec uses this old and patched version of tslib, any help from me doesn't get you anywhere anyways. They should upgrade tslib to the latest version and use tools that upstream tslib provides.

I kind of like the idea of an `module invert` filter with parameters like `x0=___` and `y0=___` for the touch point around which the inversion takes place. I might write that one, even though it's not needed when you use `linear` and ts_calibrate. There might be use cases where you don't want manual calibration and exactly such one inversion.

For future reference: IMO Libreelec should move to upstream tslib in order to get updates and new features, like multitouch. Except for such an "invert" filter, everything Libreelec needs is available in tslib's package:
* device detection
* standalone portable SDL2 graphics version of "ts_calibrate" (if needed)
* ts_uinput daemon (creates a touchscreen device with same properties as the original one, and provides tslib-filtered values) plus automatic generation of /dev/input/ts_uinput symlink.

hevilp, 2017-11-29:
Why my touchscreen got 2 devices on evtest?

I got a strange behavior:

If I change the file ts_env from
`TSLIB_TSDEVICE=$(echo 999 | evtest 2>&1 >/dev/null | awk -F':' -v TS_NAME="$TOUCHSCREEN_NAME" '$0 ~ TS_NAME {print $1; exit}')
export TSLIB_TSDEVICE
`
to 
`export TSLIB_TSDEVICE=/dev/input/event1`

I am able to touch the crosshairs und the calibration is successfull, but not calibrated, so I cannot use the screen right.

If I use the default way of libreelec, I'm not able to click the crosshair again, like on the photo above.

Why?

merge, 2017-11-29:
Like I said, I don't know Libreelec's calibration program. It's third-party and not part of tslib. It somehow is started __by__ the (also third-party) ts_uinput_touch daemon program.

Right now I can only say that these programs do something wrong in your case. The standard way would be to:
* (for your device unfortunately) you need to set TSLIB_TSDEVICE
* make sure `module linear` is in ts.conf
* run tslib's ts_calibrate
* run ts_test to verify calibration was successful
* __then__ start `ts_uinput -d -v` or the script `ts_uinput_start.sh` in order to have `/dev/input/ts_uinput` which can be used by any system in a standard way.

All those programs simply use the libts library. But you don't have these programs. You have other programs that use the library... I maintain and support what tslib ships and want to fix any problem there, but that doesn't apply in your case.

Right now you seem to be out of luck if you don't want to work on Libreelec to use an upstream version of ts_calibrate and a newer version tslib itself. I *will* work on including a generic "invert" filter for tslib here, so that in case you *can* use it, you have the filters you need.

vpeter4, 2017-11-29:
If he run ts_calibrate then he is using only tslib's functionality. But I'm sure he will be out of luck anyway.

merge, 2017-11-29:
@vpeter4 why do think so? It seems to me that if @hevilp uses Libreelec's `st1232` module_raw with parameter y_invert it should be fine. @hevilp you might even not need to calibrate at all, and remove the `module linear` from ts.conf.

vpeter4, 2017-11-29:
Is he using 7" touchscreen for Udoo board with i.MX6 SoC? No. That's why this plugin is useless for him.

merge, 2017-11-29:
ok I see. Well in that case it seems to get difficult indeed :) It's a pity though...

hevilp, 2017-11-29:
I stopped Kodi when using evtest. Seems that I cannot use libreelec, so I have to build my own debian with Kodi...., maybe with xorg-server and uses x_input_calibrate...

merge, 2017-11-29:
xinput_calibrate is quite unmaintained, old and specific to X11. Please try that and maybe it works for you.

In case you need help using tslib in Debian, I'm happy to help. At least in Debian stable, you would have to build and install tslib from the source tarball locally. It's easy though.

hevilp, 2017-11-29:
I got a fresh debian stretch running and tslib from github compiled. I think I'm missing a configuration file. I get `pi@raspberrypi:~/tslib $ ts_calibrate
ts_setup: No such file or directory
pi@raspberrypi:~/tslib $
`

merge, 2017-11-29:
No, that's most likely because tslib doesn't automatically detect your device (driver). That's a problem we can fix in Linux. I have your evtest log, and I'll look into that. Of course this doesn't help you very quickly. But simply set and export TSLIB_TSDEVICE=/dev/input/eventX to your touchscreen and start ts_calibrate again.

merge, 2017-11-29:
There are a few alternatives for you, instead of setting TSLIB_TSDEVICE, you can use `ts_calibrate -i /dev/input/eventX` directly. Or you can create a symlink `/dev/input/ts` that points to your real eventX file. ... just so you know. It all does the same: It tells tslib what input device is your touchscreen.

hevilp, 2017-11-29:
I did, my ts.config is now:
`
export TSLIB_TSDEVICE=/dev/input/event1
export TSLIB_CALIBFILE=/etc/pointercal
export TSLIB_FBDEVICE=/dev/fb0
export TSLIB_PLUGINDIR=/usr/local/lib/ts
module_raw input
module median depth=3
module dejitter delta=100
module linear
`
then:
`Couldnt load module pthres
ts_config: Success
ts_setup: Success
`

hevilp, 2017-11-29:
anyway:
`pi@raspberrypi:/usr/local/lib/ts $ ts_calibrate -i /dev/input/event1
ts_calibrate: invalid option -- 'i'`

merge, 2017-11-29:
what version of tslib do you use and how exactly did you install it? thanks.

hevilp, 2017-11-29:
A fresh install from git 30 min before 
`tslib 1.15-rc1 / libts ABI version 0 (0x000900)`

merge, 2017-11-29:
So. Don't create any "ts.config" yourself. You use `/usr/local/etc/ts.conf` which should already be installed. You __cannot__ put "export ... " in there. Leave it as it is, just make sure `module linear` is enabled.

On your commandline, do as root:

```
export TSLIB_TSDEVICE=/dev/input/event1
ts_calibrate
```

nothing more.

hevilp, 2017-11-29:
`pi@raspberrypi:/usr/local/lib/ts $ export TSLIB_TSDEVICE=/dev/input/event1
pi@raspberrypi:/usr/local/lib/ts $ ts_calibrate
Couldnt load module pthres
ts_config: Success
ts_setup: Success
pi@raspberrypi:/usr/local/lib/ts $
`
I deleted /etc/ts.conf

merge, 2017-11-29:
weird. Try to execute the ts_calibrate from your build-directory. So not just `ts_calibrate`, but /home/you/path/to/tslib_git_repo/tests/ts_calibrate instead.

In case that works, you can try to `make uninstall` and `./configure --prefix=/usr` and `make install`. But then you have to switch to using `/etc/ts.conf` and not /usr/local/...

hevilp, 2017-11-29:
Its the same, maybe I used the wrong ./configure options?

merge, 2017-11-29:
which ones did you use?

vpeter4, 2017-11-29:
Maybe also
`export TSLIB_PLUGINDIR=/usr/local/lib/ts`
or some other path where modules are installed.

hevilp, 2017-11-29:
Okay,
I uninstalled all and reinstalled it, only with ./configure without any options and now it works as root.
I get:
`root@raspberrypi:/home/pi/tslib# ts_calibrate
xres = 1184, yres = 624
Took 10 samples...
Top left : X =    0 Y =    0
Took 9 samples...
Top right : X =    0 Y =    0
Took 10 samples...
Bot right : X =    0 Y =    0
Took 8 samples...
Bot left : X =    0 Y =    0
Took 11 samples...
Center : X =    0 Y =    0
ts_calibrate: determinant is too small -- 0.000000
Calibration failed.
`

vpeter4, 2017-11-29:
Seems you don't get any useful data from touchscreen? Only zeros.

merge, 2017-11-29:
and you have `TSLIB_TSDEVICE` exported before running ts_calibrate?

hevilp, 2017-11-29:
Yep, but remember, years ago it worked fine with x_input_..... and evtest seems to work fine to0

hevilp, 2017-11-29:
yep I did, my screen got event0 and event1, If I export event0 nothing happens, If I export event1 the calibration is working, but not right. Evtest on event0 does nothing

merge, 2017-11-29:
ok. that's weird. I just now ran ts_calibrate using my laptop's touchpad as TSLIB_TSDEVICE, and it works just fine. evtest really gives you normal values on the same device that you use as TSLIB_TSDEVICE?

hevilp, 2017-11-29:
[log (2).txt](https://github.com/kergoth/tslib/files/1513850/log.2.txt)

Here is evtest with 4 points....

hevilp, 2017-11-29:
Yes, all tests are on event1. Yes Yes :)

merge, 2017-11-29:
I'm going to test ts_calibrate with an equally simple device driver like yours and try to reproduce it.

Of course you could try to build tslib with `./configure --enable-debug` and see if you get something interesting while running ts_calibrate.

hevilp, 2017-11-29:
[log (2).txt](https://github.com/kergoth/tslib/files/1514059/log.2.txt)

With debuging enabled...

merge, 2017-11-29:
thanks. nothing more interesting. I hope I can reproduce this and fix it. Otherwise I'd need an evemu recording doing 5 or 6 "taps" from you; that I can feed into ts_calibrate. Let's see.

hevilp, 2017-11-29:
[evemu.txt](https://github.com/kergoth/tslib/files/1514129/evemu.txt)
here u got it

merge, 2017-11-29:
Ok. Sorry, but I'd need a recording with like 10-15 tappings for my setup...

hevilp, 2017-11-29:
[evemu.txt](https://github.com/kergoth/tslib/files/1514174/evemu.txt)

again :)

merge, 2017-11-29:
Phew. Indeed we seem to have a problem in tslib. And as so often, affected is the old ts_read() API that is used by ts_calibrate. ts_read_mt() (for example using the program ts_print_mt) works fine. I'll look into it as soon as I can.

hevilp, 2017-11-29:
Thanks mate!

merge, 2017-11-29:
uh. I've got it. Especially for your device, the "input" module has a quirk that appearently is wrong! I won't hesitate and will do away with it, as it's really ugly and I've never experienced it to be necessary anyways. I guess it's simply fixed in Linux.

Anyways, I __think__ this is fixed for you. Please pull, build, uninstall, install and try it.

Oh, and if you want to, give me your name and email address, and I can add you to the project's THANKS file.

Even if this is fixed, you don't seem to be done :) I can promise you, I'll add an "invert" filter very soon. You seem to need that... And if you need help in generally using the tslib filtered input in Debian, please just ask.

hevilp, 2017-11-29:
So it is working now:
`root@raspberrypi:/home/pi/tslib# ts_calibrate
tslib 1.15-rc1 (library 0x000900) on Host OS: Linux, trying to open /dev/input/event1
Loading module input
Loading module pthres (pmin=1)
Loading module dejitter (delta=100)
Loading module linear
xres = 1184, yres = 624
tslib input device: vendor 0xEEF product 0x1 version 0x112 on bus 0x3
RAW---------------------> 1824 359 255 1511974009.323436
TS_READ_RAW----> x = 1824, y = 359, pressure = 255
RAW---------------------> 1825 357 255 1511974009.347417
TS_READ_RAW----> x = 1825, y = 357, pressure = 255
RAW---------------------> 1827 356 255 1511974009.363414
TS_READ_RAW----> x = 1827, y = 356, pressure = 255
RAW---------------------> 1830 354 255 1511974009.371414
TS_READ_RAW----> x = 1830, y = 354, pressure = 255
RAW---------------------> 1832 353 255 1511974009.387415
TS_READ_RAW----> x = 1832, y = 353, pressure = 255
RAW---------------------> 1833 352 255 1511974009.395413
TS_READ_RAW----> x = 1833, y = 352, pressure = 255
RAW---------------------> 1835 351 255 1511974009.411414
TS_READ_RAW----> x = 1835, y = 351, pressure = 255
RAW---------------------> 1836 349 255 1511974009.419413
TS_READ_RAW----> x = 1836, y = 349, pressure = 255
RAW---------------------> 0 0 0 1511974009.427414
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 8 samples...
Top left : X = 1831 Y =  353
RAW---------------------> 207 348 255 1511974010.283416
TS_READ_RAW----> x = 207, y = 348, pressure = 255
RAW---------------------> 206 348 255 1511974010.307406
TS_READ_RAW----> x = 206, y = 348, pressure = 255
RAW---------------------> 205 349 255 1511974010.315411
TS_READ_RAW----> x = 205, y = 349, pressure = 255
RAW---------------------> 206 352 255 1511974010.331410
TS_READ_RAW----> x = 206, y = 352, pressure = 255
RAW---------------------> 206 355 255 1511974010.339408
TS_READ_RAW----> x = 206, y = 355, pressure = 255
RAW---------------------> 207 358 255 1511974010.347410
TS_READ_RAW----> x = 207, y = 358, pressure = 255
RAW---------------------> 207 360 255 1511974010.363407
TS_READ_RAW----> x = 207, y = 360, pressure = 255
RAW---------------------> 206 361 255 1511974010.371409
TS_READ_RAW----> x = 206, y = 361, pressure = 255
RAW---------------------> 205 365 255 1511974010.387410
TS_READ_RAW----> x = 205, y = 365, pressure = 255
RAW---------------------> 0 0 0 1511974010.395410
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 9 samples...
Top right : X =  206 Y =  355
RAW---------------------> 199 1723 255 1511974011.355411
TS_READ_RAW----> x = 199, y = 1723, pressure = 255
RAW---------------------> 197 1722 255 1511974011.371403
TS_READ_RAW----> x = 197, y = 1722, pressure = 255
RAW---------------------> 195 1721 255 1511974011.387405
TS_READ_RAW----> x = 195, y = 1721, pressure = 255
RAW---------------------> 193 1719 255 1511974011.395404
TS_READ_RAW----> x = 193, y = 1719, pressure = 255
RAW---------------------> 191 1718 255 1511974011.411406
TS_READ_RAW----> x = 191, y = 1718, pressure = 255
RAW---------------------> 190 1716 255 1511974011.419404
TS_READ_RAW----> x = 190, y = 1716, pressure = 255
RAW---------------------> 188 1715 255 1511974011.435403
TS_READ_RAW----> x = 188, y = 1715, pressure = 255
RAW---------------------> 186 1714 255 1511974011.443405
TS_READ_RAW----> x = 186, y = 1714, pressure = 255
RAW---------------------> 185 1713 255 1511974011.451403
TS_READ_RAW----> x = 185, y = 1713, pressure = 255
RAW---------------------> 185 1712 255 1511974011.467405
TS_READ_RAW----> x = 185, y = 1712, pressure = 255
RAW---------------------> 184 1711 255 1511974011.475403
TS_READ_RAW----> x = 184, y = 1711, pressure = 255
RAW---------------------> 184 1708 255 1511974011.491403
TS_READ_RAW----> x = 184, y = 1708, pressure = 255
RAW---------------------> 184 1705 255 1511974011.499402
TS_READ_RAW----> x = 184, y = 1705, pressure = 255
RAW---------------------> 183 1702 255 1511974011.507405
TS_READ_RAW----> x = 183, y = 1702, pressure = 255
RAW---------------------> 180 1700 255 1511974011.523404
TS_READ_RAW----> x = 180, y = 1700, pressure = 255
RAW---------------------> 0 0 0 1511974011.531405
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 15 samples...
Bot right : X =  186 Y = 1714
RAW---------------------> 1824 1696 255 1511974012.611404
TS_READ_RAW----> x = 1824, y = 1696, pressure = 255
RAW---------------------> 1825 1696 255 1511974012.635394
TS_READ_RAW----> x = 1825, y = 1696, pressure = 255
RAW---------------------> 1826 1696 255 1511974012.643397
TS_READ_RAW----> x = 1826, y = 1696, pressure = 255
RAW---------------------> 0 0 0 1511974012.659399
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 3 samples...
Bot left : X = 1825 Y = 1696
RAW---------------------> 1013 1100 255 1511974015.555395
TS_READ_RAW----> x = 1013, y = 1100, pressure = 255
RAW---------------------> 1014 1101 255 1511974015.579386
TS_READ_RAW----> x = 1014, y = 1101, pressure = 255
RAW---------------------> 1014 1102 255 1511974015.595384
TS_READ_RAW----> x = 1014, y = 1102, pressure = 255
RAW---------------------> 1015 1102 255 1511974015.603384
TS_READ_RAW----> x = 1015, y = 1102, pressure = 255
RAW---------------------> 0 0 0 1511974015.619385
TS_READ_RAW----> x = 0, y = 0, pressure = 0
Took 4 samples...
Center : X = 1014 Y = 1101
1271.112671 -0.664242 -0.006356
-94.339966 0.002357 0.387003
Calibration constants: 83303640 -43531 -416 -6182664 154 25362 65536
`

merge, 2017-11-29:
Alright. Now how is things looking when you run `ts_test` or `ts_test_mt`?. Maybe you don't need an invert filter after all. Still, I added one because these are the things I want to support in tslib; and there might be cases where inversion is needed, but the "linear" filter cannot be used. It needs at least some testing for the release but it's simple enough.

(One hint for using tslib in Debian is https://github.com/merge/xf86-input-tslib )

hevilp, 2017-11-29:
I will test it soon?
You got the filter already in the code?

merge, 2017-11-29:
Yes I do

Am 29. November 2017 18:19:24 MEZ schrieb hevilp <notifications@github.com>:
>I will test it soon?
>You got the filter already in the code?

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

hevilp, 2017-11-29:
How do I use it?

merge, 2017-11-29:
it should be documented in the README.md. what's unclear?

Am 29. November 2017 18:22:10 MEZ schrieb hevilp <notifications@github.com>:
>How do I use it?

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

hevilp, 2017-11-29:
We are speaking about the new filter? ts_test is working very good.

merge, 2017-11-29:
yes. well, then you have found your configuration. nice.

Am 29. November 2017 18:43:52 MEZ schrieb hevilp <notifications@github.com>:
>We are speaking about the new filter? ts_test is working very good.

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

hevilp, 2017-11-29:
I guess, till libreelec doesnt use the new sources, I'm not able to use it, since the "old" version is incompatible with my setup, right? I tried to use the pointercal in libreelec with no success -> not calibrated...

[6 more replies]

closed 2017-11-30 by merge