libts/tslib #119: Raw ts_* working - inside application not

hevilp, 2017-11-30:
Hi agai,

as you know, ts_calibrate and ts_test working like a charm.
I installed a fresh Kodi. (without x-server).

I start Kodi from commandline in framebuffer. The x-axis is still inverted, why the hell? In ts_test all is working fine.

merge, 2017-11-30:
Why the hell? That's probably because Kodi doesn't know how to use tslib's input samples. If it would, it would directly implement tslib's API and link against tslib and you wouldn't have to install tslib in the first place, because it would have been automatically there.

Have you read the tslib README documentation about "using tslib in your environment"? The way to go for you is, simply starting "ts_uinput -d -v" during startup, before you start kodi. The details depend on your setup.

merge, 2017-11-30:
ts_uinput is a program that uses tslib, just like "ts_calibrate". So, if you set TSLIB_TSDEVICE for ts_calibrate, also do so for ts_uinput.

hevilp, 2017-11-30:
I tried that, it creates a virtual device /dev/input/event3. I need to tell Kodi to use the new virtual device I think

merge, 2017-11-30:
You are exactly right. You have to refer to Kodi's documentation for this part.

merge, 2017-12-01:
I don't think this is a tslib issue, so I'll leave it open for a little while. If you don't have any problems with tslib, please close it. Otherwise I'll do so soon. Thanks for your help so far!

hevilp, 2017-12-01:
I tried with OSMC and raspbian. I cannot get it working in Kodi. the calibration is fine, ts_test is working (except libreelec) but in Kodi its not calibrated and inverted. 

Libreelec uses ts_uinput and maybe I am waiting till you release 1.15 final and they pull it in their build.

thanks

merge, 2017-12-01:
I don't think Libreelec uses ts_uinput. And they don't update tslib. They use a patched version of tslib-1.1 I think and stay at it. You'd have to work on Libreelec and change all their calibration behaviour they have right now. (i.e. drop their calibration program and configure the system to just run tslib's "ts_calibrate" on startup, before Kodi starts, in case there is no "pointercal" file; and so on...)

Your problem seems to be to tell Kodi to use one specific input event device. One can do so in Qt, in X11, in Wayland or whatever Kodi is. That's no tslib problem actually. ts_uinput only happens to create the input event device for you. Could be any driver though. That's why, in case there no other problem, I'd close this issue soon (which should not stop you from further replying and discussion of course!!)

hevilp, 2017-12-01:
I missunderstand me, I do not use their calibration programm. I use ts_calibrate, but I'm pretty sure, they use ts_uinput.
Please take a look in 
https://github.com/LibreELEC/LibreELEC.tv/tree/master/packages/addons/service/touchscreen/source/bin

this is the addon we are talking about

merge, 2017-12-01:
no they don't. Their daemon driver program, as you can see, is called "ts_uinput_touch" and is added in a patch against tslib-1.1 here: https://github.com/LibreELEC/LibreELEC.tv/tree/master/packages/addons/addon-depends/tslib/patches

And this "ts_uinput_touch" does other things than tslib's "ts_uinput". "ts_uinput" is a _userspace input event device driver_ for tslib. It creates a (second) touchscreen device called "ts_uinput", that looks almost identical to your original one, and gives you the filtered x/y... values. It can be used by any environment that currently uses your original touchscreen events. Just switch over to your new touchscreen :)

hevilp, 2017-12-01:
Okay, 1 Question, can you fork and patch the 1.15 to use their daemon? If not I will try to patch the 1.15 to use their ts_uinput_touch...

merge, 2017-12-01:
I won't fork tslib. I maintain the project and will keep trying to make it work for anybody. Not only for Kodi. Everything Kodi needs is already there in tslib's package.

If you start working on it, I would refactor things at Libreelec to use tslib's ts_uinput daemon *and* ts_calibrate too. It surely is possible and would make the "addon" nicely maintainable without any patches. You'd get all tslib updates by simply bumping the version number.

merge, 2017-12-01:
Otherwise you would need to maintain your fork (ts_uinput_touch against tslib 1.15) forever again, or you have the same situation like now, that tslib stays at 1.15 forever.

hevilp, 2017-12-01:
You are right. It will be to high for me. I'm just a user ^^.
I can't even make it work in a normal distribution like debian stretch....

hevilp, 2017-12-01:
I will try to create another Addon for Kodi, which uses your tslib with ts_uinput...

merge, 2017-12-01:
Why not? Debian uses X11. Install xf86-input-tslib from https://github.com/merge/xf86-input-tslib or just run tslib's script `ts_uinput_start.sh` and write something like `Device=/dev/input/ts_uinput` in the correct xorg config file.

hevilp, 2017-12-01:
The trick is not to use X11, it all runs in the framebuffer without X11...

merge, 2017-12-01:
ok. If you have something you want me to review, feel free to reply here or write to the mailing list.

closed 2017-12-01 by merge

vpeter4, 2017-12-01:
If time permits I will update tslib in libreelec. But no promises.
Continuation on libreelec forum...