libts/tslib #120: ts_uinput without pressure event from touchscreen

vpeter4, 2017-12-04:
I'm using touchscreen on Udoo Quad board
https://shop.udoo.org/eu/quad-dual/udoo-quad.html
https://shop.udoo.org/eu/video-kit-15-6-lvds-for-quaddual.html
but I think other displays can have same problem.

```
LibreELEC:~ # evtest /dev/input/event2
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x596 product 0x1 version 0x410
Input device name: "3M 3M USB Touchscreen - EX II"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value   6252
      Min        0
      Max    16384
    Event code 1 (ABS_Y)
      Value   3993
      Min        0
      Max    16384
Properties:
Testing ... (interrupt to exit)
Event: time 1512410518.959193, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1512410518.959193, type 3 (EV_ABS), code 0 (ABS_X), value 4213
Event: time 1512410518.959193, type 3 (EV_ABS), code 1 (ABS_Y), value 2989
Event: time 1512410518.959193, -------------- SYN_REPORT ------------
Event: time 1512410518.961182, type 3 (EV_ABS), code 0 (ABS_X), value 4216
Event: time 1512410518.961182, type 3 (EV_ABS), code 1 (ABS_Y), value 2984
Event: time 1512410518.961182, -------------- SYN_REPORT ------------
Event: time 1512410518.963175, type 3 (EV_ABS), code 0 (ABS_X), value 4218
Event: time 1512410518.963175, type 3 (EV_ABS), code 1 (ABS_Y), value 2983
Event: time 1512410518.991192, -------------- SYN_REPORT ------------
Event: time 1512410518.993170, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1512410518.993170, -------------- SYN_REPORT ------------
```

As visible it doesn't support pressure event. My ts_uinput_touch daemon supports pressure events but I think this is not really important. More important is that it sends button touch event correctly and Kodi reacts like it should.

```
LibreELEC:~ # evtest /dev/input/event3
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x34a5 product 0x67b8 version 0x1
Input device name: "tslib to uinput daemon"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value    197
      Min        0
      Max     1360
    Event code 1 (ABS_Y)
      Value    371
      Min        0
      Max      768
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)
Event: time 1512410405.472672, type 3 (EV_ABS), code 0 (ABS_X), value 89
Event: time 1512410405.472672, type 3 (EV_ABS), code 1 (ABS_Y), value 122
Event: time 1512410405.472672, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 255
Event: time 1512410405.472672, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1512410405.472672, -------------- SYN_REPORT ------------
Event: time 1512410405.486680, type 3 (EV_ABS), code 1 (ABS_Y), value 121
Event: time 1512410405.486680, -------------- SYN_REPORT ------------
Event: time 1512410405.494672, type 3 (EV_ABS), code 0 (ABS_X), value 88
Event: time 1512410405.494672, -------------- SYN_REPORT ------------
Event: time 1512410405.508694, type 3 (EV_ABS), code 0 (ABS_X), value 89
Event: time 1512410405.508694, -------------- SYN_REPORT ------------
Event: time 1512410405.516665, type 3 (EV_ABS), code 1 (ABS_Y), value 122
Event: time 1512410405.584688, -------------- SYN_REPORT ------------
Event: time 1512410405.584738, type 3 (EV_ABS), code 24 (ABS_PRESSURE), value 0
Event: time 1512410405.584738, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1512410405.584738, -------------- SYN_REPORT ------------
```

But ts_uinput daemon is not sending button touch event correctly. Probably because it doesn't get pressure events from touchscreen?

```
LibreELEC:~ # evtest /dev/input/event3
Input driver version is 1.0.1
Input device ID: bus 0x6 vendor 0x0 product 0x0 version 0x0
Input device name: "ts_uinput"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max     1359
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max      767
Properties:
Testing ... (interrupt to exit)
Event: time 1512411351.653637, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1512411351.653637, type 3 (EV_ABS), code 0 (ABS_X), value 65
Event: time 1512411351.653637, type 3 (EV_ABS), code 1 (ABS_Y), value 120
Event: time 1512411351.653637, -------------- SYN_REPORT ------------
Event: time 1512411351.671282, type 3 (EV_ABS), code 0 (ABS_X), value 64
Event: time 1512411351.671282, -------------- SYN_REPORT ------------
Event: time 1512411351.675308, type 3 (EV_ABS), code 1 (ABS_Y), value 121
Event: time 1512411351.675308, -------------- SYN_REPORT ------------
Event: time 1512411351.683307, type 3 (EV_ABS), code 1 (ABS_Y), value 122
Event: time 1512411351.683307, -------------- SYN_REPORT ------------
Event: time 1512411351.689299, type 3 (EV_ABS), code 1 (ABS_Y), value 124
Event: time 1512411351.689299, -------------- SYN_REPORT ------------
Event: time 1512411351.691309, type 3 (EV_ABS), code 0 (ABS_X), value 65
Event: time 1512411351.691309, type 3 (EV_ABS), code 1 (ABS_Y), value 126
Event: time 1512411351.691309, -------------- SYN_REPORT ------------
Event: time 1512411351.693302, type 3 (EV_ABS), code 0 (ABS_X), value 71
Event: time 1512411351.693302, type 3 (EV_ABS), code 1 (ABS_Y), value 134
Event: time 1512411351.693302, -------------- SYN_REPORT ------------
...........
Event: time 1512411362.851355, type 3 (EV_ABS), code 0 (ABS_X), value 179
Event: time 1512411362.851355, type 3 (EV_ABS), code 1 (ABS_Y), value 315
Event: time 1512411362.851355, -------------- SYN_REPORT ------------
```

**..... missing BTN_TOUCH with value 0**

merge, 2017-12-05:
Thanks for reporting! If you can give me an evemu-recording of your __original__ device (of a few "taps"), I can test myself. That's be great. Otherwise I'll try to fix it anyways, requiring you to test...

vpeter4, 2017-12-05:
Will do.

vpeter4, 2017-12-05:
Directly from touchscreen: http://sprunge.us/FVEj
```
LibreELEC:~ # evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      RCMCU iPazzPort
/dev/input/event1:      RCMCU iPazzPort
/dev/input/event2:      3M 3M USB Touchscreen - EX II   <<<< this one
Select the device event number [0-2]:
```

merge, 2017-12-05:
I think that's not a complete recording :( The descriptors on top are missing..

Am 5. Dezember 2017 18:26:23 MEZ schrieb "Peter V." <notifications@github.com>:
> Directly from touchscreen: http://sprunge.us/FVEj
> ```
> LibreELEC:~ # evtest
> No device specified, trying to scan all of /dev/input/event*
> Available devices:
> /dev/input/event0:      RCMCU iPazzPort [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

vpeter4, 2017-12-05:
Seems I was using some prehistoric evemu :(

I think this one should be better: http://sprunge.us/AFQS

merge, 2017-12-05:
great. thanks

Am 5. Dezember 2017 19:25:37 MEZ schrieb "Peter V." <notifications@github.com>:
>Seems I was using some prehistoric evemu :(
>
>I think this one should be better: http://sprunge.us/AFQS

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

merge, 2017-12-06:
Sorry, but now your recording isn't available anymore :(

I  __think__ that I have the fix and will push it, but please, if you have the time, re-opload this recording for verification. thanks!

vpeter4, 2017-12-06:
I think the problem still exists after quick test.

evemu record: http://vpeter.libreelec.tv/touchscreen/3M.3M.USB.Touchscreen.-.EX.II.evemu.record.txt

merge, 2017-12-06:
looks ok to me: recording ts_uinput's device now, when TSLIB_TSDEVICE is your sample, is:

```
# EVEMU 1.3
# Kernel: 4.14.0-00002-g22faf2dbad1a
# DMI: dmi:bvnFUJITSU//PhoenixTechnologiesLtd.:bvrVersion1.11:bd01/13/2015:svnFUJITSU:pnCELSIUSH730:pvr10601583184:rvnFUJITSU:rnFJNB274:rvrJ3:cvnFUJITSU:ct10:cvrCELSIUSH730:
# Input device name: "ts_uinput"
# Input device ID: bus 0x06 vendor 0000 product 0000 version 0000
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 330 (BTN_TOUCH)
#   Event type 3 (EV_ABS)
#     Event code 0 (ABS_X)
#       Value    -7900
#       Min          0
#       Max       1919
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 1 (ABS_Y)
#       Value    -5288
#       Min          0
#       Max       1079
#       Fuzz         0
#       Flat         0
#       Resolution   0
# Properties:
N: ts_uinput
I: 0006 0000 0000 0000
P: 00 00 00 00 00 00 00 00
B: 00 0b 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 04 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 02 00 00 00 00 00 00 00 00
B: 03 03 00 00 00 00 00 00 00
B: 04 00 00 00 00 00 00 00 00
B: 05 00 00 00 00 00 00 00 00
B: 11 00 00 00 00 00 00 00 00
B: 12 00 00 00 00 00 00 00 00
B: 14 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
A: 00 0 1919 0 0 0
[... 142 lines cut ...]
E: 4.826058 0003 0001 -5126	# EV_ABS / ABS_Y                -5126
E: 4.826058 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +4ms
E: 4.828025 0003 0000 -7769	# EV_ABS / ABS_X                -7769
E: 4.828025 0003 0001 -5133	# EV_ABS / ABS_Y                -5133
E: 4.828025 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +2ms
E: 4.829981 0003 0000 -7771	# EV_ABS / ABS_X                -7771
E: 4.829981 0003 0001 -5140	# EV_ABS / ABS_Y                -5140
E: 4.829981 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +1ms
E: 4.833920 0003 0000 -7767	# EV_ABS / ABS_X                -7767
E: 4.833920 0003 0001 -5150	# EV_ABS / ABS_Y                -5150
E: 4.833920 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +4ms
E: 4.835838 0003 0000 -7757	# EV_ABS / ABS_X                -7757
E: 4.835838 0003 0001 -5162	# EV_ABS / ABS_Y                -5162
E: 4.835838 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +2ms
E: 4.837767 0003 0001 -5174	# EV_ABS / ABS_Y                -5174
E: 4.837767 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +2ms
E: 4.841725 0003 0000 -7766	# EV_ABS / ABS_X                -7766
E: 4.841725 0003 0001 -5189	# EV_ABS / ABS_Y                -5189
E: 4.841725 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +4ms
E: 4.843689 0003 0000 -7900	# EV_ABS / ABS_X                -7900
E: 4.843689 0003 0001 -5288	# EV_ABS / ABS_Y                -5288
E: 4.843689 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +2ms
E: 4.845636 0001 014a 0000	# EV_KEY / BTN_TOUCH            0
E: 4.845636 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +2ms
```

merge, 2017-12-06:
Maybe you can test the current state on a real device. But if you still can't tap and don't get BTN_TOUCH correctly, please re-open the issue!

thanks for your help

vpeter4, 2017-12-06:
I did test on real device and BTN_TOUCH just got stuck with value 1. But maybe it was too early in the day when I was still sleeping. Will do later today more tests.

vpeter4, 2017-12-06:
The problem I had is using 
`module variance delta=30`
from old config. With using this module BTN_TOUCH got stuck with value 1.

But without this module BTN_TOUCH is working fine.
Seems there was nothing wrong with ts_uinput at the end but problematic is variance module?

But Kodi is not happy yet:
```
16:00:13.102 T:3016482816   DEBUG: ProcessMouse: unknown mouse command 57605
16:00:13.102 T:3016482816   DEBUG: ProcessMouse: trying mouse action mousedrag
16:00:13.103 T:3016482816   DEBUG: ProcessMouse: unknown mouse command 57606
16:00:15.247 T:3016482816   DEBUG: ProcessMouse: unknown mouse command 57605
16:00:15.247 T:3016482816   DEBUG: ProcessMouse: trying mouse action mousedrag
16:00:15.248 T:3016482816   DEBUG: ProcessMouse: unknown mouse command 57606
16:00:15.383 T:3016482816   DEBUG: ProcessMouse: unknown mouse command 57605
16:00:15.383 T:3016482816   DEBUG: ProcessMouse: trying mouse action mousedrag
16:00:15.383 T:3016482816   DEBUG: ProcessMouse: unknown mouse command 57606
```
As indicated Kodi recognizes my touchscreen as ordinary mouse. Which even doesn't work :(
I will investigate more in next days and compare with my daemon where I had this working. I think I also have one smaller touchscreen with dual touch.

But now I'm wondering with the state of touch support in Kodi. At least in v17. Maybe multitouch will not even work. Sadly I can't test with Kodi 18 because I don't have any regular touchscreen (hdmi + usb) to test it on x86 based system.

merge, 2017-12-07:
This should be verified to work with your provided evemu-recording. I currently don't have much time, but it should be done before the release:

* use only module_raw input
* evemu record ts_uinput against the evemu-sample
* compare in mtview or the like

[3M.3M.USB.Touchscreen.-.EX.II.evemu.record.txt](https://github.com/kergoth/tslib/files/1538762/3M.3M.USB.Touchscreen.-.EX.II.evemu.record.txt)

vpeter4, 2017-12-07:
Maybe Kodi would not even work with my touchscreen but in this case I'm expecting from ts_uinput daemon to do whatever is necessary to emulate at least mouse for example. This is what my original daemon is capable and I'm expecting the same from this new daemon. If not then it is useless for my purpose without some extra patching.

There is also one extra feature I have in my daemon: showing cursor only when tapped on screen. When not tapped cursor is moved outside of visible area: https://github.com/vpeter4/tslib/blob/a5f00518cfe38002c47895424eda26746711a0da/tests/ts_uinput_touch.c#L617:L619 But this is a feature after everything else will work without a hitch ;)

vpeter4, 2017-12-27:
I did look into why ts_uinput doesn't like to work with kodi. With this changes https://pastebin.com/H9fk6GkL it does. It adds ABS_PRESSURE event and send BTN_TOUCH event after pressure. I'm using only one finger touch. I'm not sure if kodi even supports multi touch events.

merge, 2017-12-28:
Thanks for helping! I still think that adding pressure to the input device is not correct here. The original device doesn't need it and so don't we. input-raw creates pressure for internal use, but it's output should be usable with the same properties as the original device.

But doesn't your fix really need the following hunk?

```
@@ -419,6 +427,7 @@ static int setup_uinput(struct data_t *d
        if (ioctl(data->fd_uinput, UI_SET_EVBIT, EV_KEY) < 0 ||
            ioctl(data->fd_uinput, UI_SET_EVBIT, EV_SYN) < 0 ||
            ioctl(data->fd_uinput, UI_SET_KEYBIT, BTN_TOUCH) < 0 ||
+           ioctl(data->fd_uinput, UI_SET_ABSBIT, ABS_PRESSURE) < 0 ||
            ioctl(data->fd_uinput, UI_SET_EVBIT, EV_ABS) < 0) {
                perror("ioctl");
                goto err;
@@ -428,6 +437,9 @@ static int setup_uinput(struct data_t *d
        snprintf(uidev.name, UINPUT_MAX_NAME_SIZE, "%s", data->uinput_name);
        uidev.id.bustype = BUS_VIRTUAL;

+  uidev.absmin[ABS_PRESSURE] = 0;
+  uidev.absmax[ABS_PRESSURE] = 255;
+
        memset(bit, 0, sizeof(bit));

        /* get info on input device and copy setting over to uinput device */

```

thanks

vpeter4, 2017-12-28:
Seems with yesterday's tests I missed one combination of two patches because now it works just with moving BTN_TOUCH. So the ABS_PRESSURE is not needed after all as you suspect :)

merge, 2017-12-28:
I'd like some feedback too, hence -rc3 now. I'll test for regressions next week and if ok, do the release. Thanks!

vpeter4, 2017-12-28:
I do have another system with LVDS touchscreen but seems it is broken somehow :(

merge, 2017-12-28:
You seem to have a problem with tslib, or your hardware?

evemu tests look ok. I think I'll get other feedback too, and do regression tests myself. Please test if you can, but no stress; we'll cover testing this change and do a release early next year if ok.

Am 28. Dezember 2017 18:58:52 MEZ schrieb "Peter V." <notifications@github.com>:
>I do have another system with LVDS touchscreen but seems it is broken
>somehow :(

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

vpeter4, 2017-12-28:
No, no, tslib is ok - hardware is faulty (it doesn't boot) :)

Fred-cn, 2018-01-06:
My Qt5 app didn't receive any touch event after using this ts_uinput tool.
My configure is in the following.
TSLIB_TSDEVICE=/dev/input/event4 -->this is my touchscreen node.(ts_calibrate work fine.)
QT_QPA_GENERIC_PLUGINS=evdevtouch:/dev/input/event5-->this is virtual node that generated by ts_uinput.(When I change this node to /dev/input/event4, It work fine too,But received point was not calibrated.)

vpeter4, 2018-01-06:
I finally got my other display with dual touch up and running. It is https://shop.udoo.org/eu/catalog/product/view/id/23/s/video-kit-7-touch-for-quaddual/category/3/
It uses st1232 kernel module and doesn't need calibration. Kodi doesn't like it (no reaction on presses).
Probably because it doesn't support BTN_TOUCH as we already figure out it is needed.

evemu: https://pastebin.com/q9P3M8SG
```
# EVEMU 1.3
# Kernel: 3.14.79
# Input device name: "st1232-touchscreen"
# Input device ID: bus 0x18 vendor 0000 product 0000 version 0000
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#   Event type 3 (EV_ABS)
#     Event code 48 (ABS_MT_TOUCH_MAJOR)
#       Value        0
#       Min          0
#       Max        255
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 53 (ABS_MT_POSITION_X)
#       Value        0
#       Min          0
#       Max        799
#       Fuzz         0
#       Flat         0
#       Resolution   0
#     Event code 54 (ABS_MT_POSITION_Y)
#       Value        0
#       Min          0
#       Max        479
#       Fuzz         0
#       Flat         0
#       Resolution   0
# Properties:
N: st1232-touchscreen
I: 0018 0000 0000 0000
P: 00 00 00 00 00 00 00 00
B: 00 0b 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 02 00 00 00 00 00 00 00 00
B: 03 00 00 00 00 00 00 61 00
[... 319 lines cut ...]
E: 2.234541 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 2.250523 0003 0035 0070	# EV_ABS / ABS_MT_POSITION_X    70
E: 2.250523 0003 0036 0404	# EV_ABS / ABS_MT_POSITION_Y    404
E: 2.250523 0003 0030 0059	# EV_ABS / ABS_MT_TOUCH_MAJOR   59
E: 2.250523 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
E: 4.161155 0003 0035 0075	# EV_ABS / ABS_MT_POSITION_X    75
E: 4.161155 0003 0036 0421	# EV_ABS / ABS_MT_POSITION_Y    421
E: 4.161155 0003 0030 0044	# EV_ABS / ABS_MT_TOUCH_MAJOR   44
E: 4.161155 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +1911ms
E: 4.169227 0003 0035 0075	# EV_ABS / ABS_MT_POSITION_X    75
E: 4.169227 0003 0036 0421	# EV_ABS / ABS_MT_POSITION_Y    421
E: 4.169227 0003 0030 0042	# EV_ABS / ABS_MT_TOUCH_MAJOR   42
E: 4.169227 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +8ms
E: 4.184769 0003 0035 0075	# EV_ABS / ABS_MT_POSITION_X    75
E: 4.184769 0003 0036 0421	# EV_ABS / ABS_MT_POSITION_Y    421
E: 4.184769 0003 0030 0040	# EV_ABS / ABS_MT_TOUCH_MAJOR   40
E: 4.184769 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +15ms
E: 4.200337 0003 0035 0076	# EV_ABS / ABS_MT_POSITION_X    76
E: 4.200337 0003 0036 0415	# EV_ABS / ABS_MT_POSITION_Y    415
E: 4.200337 0003 0030 0032	# EV_ABS / ABS_MT_TOUCH_MAJOR   32
E: 4.200337 0000 0000 0000	# ------------ SYN_REPORT (0) ---------- +16ms
```

My view on ts_uinput daemon is that it should support such cases just like my original daemon does.
If this is not possible by default then it should be configured with some arguments.

merge, 2018-01-08:
this is a type A multitouch device and we indeed handle this in a wrong way. I haven't tested it myself until now. Thanks for your sample. We should keep that sample and fix it.

merge, 2018-01-09:
Ok. ts_uinput definitely cannot handle multitouch type A devices (with SYN_MT_REPORT). That would be a seperate issue.

For all other devices, please test the current master state if you can.

vpeter4, 2018-01-09:
Sadly Kodi doesn't like tslib current master (https://github.com/kergoth/tslib/commit/fbeabc67146095c5da787c5d849b9f14bbe0d6a4) with my single touch screen (3M 3M USB Touchscreen - EX II). I can select some item on main menu but when I tap on it it doesn't select it. I assume because BTN_TOUCH event is first and then X/Y coordinates. When first is X/Y coordinate and then BTN_TOUCH it works beautifully. No idea which fault is this (maybe Kodi's?).

merge, 2018-01-10:
At least that ordering is something that should be easily done ts_uinput, just pushed to master.

vpeter4, 2018-01-10:
I think I'm not the right person to report such bugs because I'm testing only with Kodi. And after looking used code in Kodi 17.6 https://github.com/xbmc/xbmc/blob/Krypton/xbmc/input/linux/LinuxInputDevices.cpp 
(and kodi master is very similar but I can't test that https://github.com/xbmc/xbmc/blob/master/xbmc/platform/linux/input/LinuxInputDevices.cpp) it needs coordinates first and then BTN_TOUCH event which eventually fires left click (because it simulates mouse). If BTN_TOUCH event is send before X/Y then it fires click at coordinate 0,0. And this doesn't work obviously :)

I will just patch my own build with this change - it will be easier.

commit 69757330 (2017-12-06): input-raw: mark BTN_TOUCH code valid

In case we have *only* BTN_TOUCH in a sync-frame, we had not marked the
frame as valid and thus missed pen-up for device that *don't* report
pressure or tracking-id or pen-up in any other way. So fix this.

Also, this removes the obsolete handling of BTN_LEFT for ts_read_mt().
It did nothing, and driver's shouldn't use it anyways.

Fixes #120
commit cafd5bb4 (2019-01-23): input-raw / ts_uinput: fix MT type A device detection and conversion

ts_uinput always output a type B device (if MT). This fixes input-raw's
type A detection and data-generation for type B data.

That said, we do not go as far as libmtdev (http://bitmath.org/code/mtdev/)
goes with A to B conversion. For us, samples that come in first (in a frame)
are slot 0, and so on. No real tracking of touch points using distance
calculations based on previous states. Still, this should result in a totally
usable input stream for most users.

Also, ts_uinput now actually does what it claims to do. If it de [... cut]
closed 2019-01-23 by merge