libts/tslib #121: CrossCompile-Error

hevilp, 2017-12-06:
Hi,

I try to crosscompile from 16.04 Ubuntu LTS for Raspberry Pi 3 and get 
rpl_malloc errors:

Source: https://github.com/kergoth/tslib/releases/download/1.15-rc2/tslib-1.15-rc2.tar.xz

https://pastebin.com/1VB965eq

Compiling on Rpi3 directly is successfull.

hevilp, 2017-12-07:
Still the error:

https://pastebin.com/sSdS9ETc

merge, 2017-12-07:
Not sure if it's because of our alloca(), but have you tried to `export ac_cv_func_malloc_0_nonnull=yes` before ./configure ?

See https://stackoverflow.com/a/9243043/6435680

hevilp, 2017-12-07:
Solved it!

vpeter4, 2017-12-07:
@hevilp: seems you missed my comments at https://github.com/LibreELEC/LibreELEC.tv/pull/2281#issuecomment-348432528

closed 2017-12-07 by merge