libts/tslib #122: Multiple Touchscreen Devices on USB hub

MonteShaffer, 2017-12-11:
Hi, tslib is good stuff thank you.

On a Debian platform, I have a USB hub with say 3+ touchscreens (same model).  Does `ts_calibrate` allow me to identify which device I am working on.  I would want to be able to identify and calibrate each of the 3+ touchscreens so when I make a call, I know which device.  That is, each call would include an extra parameter (deviceID).  Isn't unique device labeling <Device Serial> a standard for the USB protocol?

I saw a closed comment related to [https://stackoverflow.com/questions/11437435/] and wanted to know if this is the same question, and what the status is.

Anyway, I am just now entering the domain of this issue, and want to know if I understand the problem correctly.  Currently, with tslib, I can not do the following:

- Run "Identify" to map the Device Serial to some simple Device Number for all available touchscreens in USB pool
- For each DeviceNumber, calibrate the given device and create the required file
- For a given display task, refer to the DeviceNumber with the task.  For an event (touch listener) have it be bound to the DeviceNumber

I am a newbie in this space, but a fast learning, so please let me know if the above is currently possible.  If not, is it on the roadmap, and what will it take to get there.

We are needing multiple touchscreens in the development of a solution to help those that suffer from Parkinson's Disease.

Cheers!

merge, 2017-12-11:
You can use tslib (ts_calibrate or any program provided) with a specified device, see our docs:
* export `TSLIB_TSDEVICE=/dev/input/eventX` for it. OR
* create a symlink `/dev/input/ts` pointing to the touchscreen.

So simply re-set that before you execute ts_calibrate.

The "identify" part you actually have to do on your own. We don't include a tool that "lists all touchscreen devices". We only "fall back" to auto-detect the (first) touchscreen device if you specify nothing. One way to do it is to identify it in `/proc/bus/input/devices` (INPUT_PROP_DIRECT should be set for touchscreens). Also, please have a look if the [linux console project](https://sourceforge.net/projects/linuxconsole/) has a tool you could use. "List touchscreen devices" feels generic enough that it hasn't have to be done by one of our programs. Do you think we should create a tool for that?

As for "DeviceNumber", you always simply have the driver's event file. But in your case, I'd set up symlinks during startup. event device file numbers do *not* stay the same over time.

And what do you mean by display task? So you write code against our library and have an eventloop in C or C++?

MonteShaffer, 2017-12-11:
Thanks merge for the response, that should get me started.

The symlink hint will be very helpful, and I will dig into this over the next few weeks.

Good question about "display task" - I will let you know more when I am in the middle of the problem.

When I figure out the "identify", loop and "symlink" - I will share the code. I believe it would be useful for  the community.

merge, 2017-12-21:
This doesn't seem to be an issue for us now. I'll go ahead and close it, but please feel free to reply anyways or reopen if you think there is an issue for us to work on.

closed 2017-12-21 by merge