libts/tslib #125: ts_uinput not detecting touch events from /dev/input/event0

barteksc, 2018-03-21:
Hello,

I'am using Raspberry Pi Zero with Waveshare WS170120 HDMI display. Resistive touch layer is connecter through USB and works out of the box with `/dev/input/event0`.
I didn't find `ts_uinput` executable in Raspbian repos, so I've built tslib manually from cloned git repo:
```
$ ./autogen.sh
$ ./configure
$ make
# make install
```

I'm creating GUI through framebuffer, so X server is not available.
To calibrate touch screen I've defined global environment variables:
```
TSLIB_TSDEVICE=/dev/input/event0
TSLIB_FBDEVICE=/dev/fb0
```
and then I've used `# ts_calibrate` and `# ts_test` - it was working without any problems.
Running `# ts_uinput -v` creates `/dev/input/event1` device and I see following output in terminal when touching the screen:
```
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: sample 0:  x	y	slot	tracking_id
...
```

The problem is that when I use `evtest /dev/input/event1` I see some info output and app is waiting for touch events, but when I touch the screen nothing happens. When I run `evtest /dev/input/event0`, it is working as expected 
```
Event: time 1521591643.421618, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 81
Event: time 1521591643.421618, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 2001
Event: time 1521591643.421618, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 2716
Event: time 1521591643.421618, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
...
```

I think that I'm missing something, but I'm out of ideas. Thank you in advance for any help.

merge, 2018-03-21:
Hello,

how does your ts.conf look?

barteksc, 2018-03-21:
I use default config from `libts-bin` Raspbian's package:
```
module_raw input

module pthres pmin=1
module variance delta=30
module dejitter delta=100
module linear
```

I've also noticed, that on first run of `ts_uinput -v`, just after system reboot, one gesture is recognized. When I want to do second gesture, it stops working.
```
...
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	348	194	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	348	194	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	348	194	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	348	194	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	349	194	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	350	193	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	351	193	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	352	192	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	352	191	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	356	189	0	0
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: 	361	186	0	-1
--- second gesture ---
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: sample 0:  x	y	slot	tracking_id
ts_uinput: sample 0:  x	y	slot	tracking_id
...
```

merge, 2018-03-21:
Ok, that's strange indeed. Please definitely try removing all filters you dont need from ts.conf, especially variance.

So please try having only "modul_raw input" and "linear" (to apply your calibration).

Also, what version of tslib do you use?

thanks

Am 21. März 2018 19:16:26 MEZ schrieb barteksc <notifications@github.com>:
> I use default config from `libts-bin` Raspbian's package:
> ```
> module_raw input
>
> module pthres pmin=1
> module variance delta=30
> module dejitter delta=100
> module linear
> ```
>
> I've also noticed, that on [...]

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

barteksc, 2018-03-21:
My version is `tslib 1.15 / libts ABI version 0 (0x000900)`, I've built it from `master` branch yesterday.
Unfortunately, config changes didn't help.

merge, 2018-03-21:
Ok. Can you do one recording with evemu-record (evemu raspbian package I think) of your original touchscreen, doing 2 or 3 short gestures? You can upload a file here I think.

(evemu-record is almost the same as evtest, but I can replay it and reproduce your problem that way, you know)

I know I keep you busy. Please ask if something's not clear.

Am 21. März 2018 19:36:27 MEZ schrieb barteksc <notifications@github.com>:
>My version is `tslib 1.15 / libts ABI version 0 (0x000900)`, I've built
>it from `master` branch yesterday.
>Unfortunately, config changes didn't help.

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

barteksc, 2018-03-21:
I've edited my previous comment, it started working after config changes.
Maybe I should have create new comment, so you would be notified about my update.
Thank you again.

merge, 2018-03-21:
Good. If you feel like it, edit the title to "bad rapbian default config" or something.

Am 21. März 2018 20:45:04 MEZ schrieb barteksc <notifications@github.com>:
>I've edited my previous comment, it started working after config
>changes.
>Maybe I should have create new comment, so you would be notified about
>my update.
>Thank you again.

-- 
Martin Kepplinger
http://martinkepplinger.com
sent from mobile

merge, 2018-03-22:
actually, from what I see at https://archive.raspbian.org/raspbian/pool/main/t/tslib/ they use Debian's config, which is our upstream config, which is fine.

Do you maybe have the "tsconf" raspbian package installed? That is old and should actually (automatically) be removed when installing tslib...